WildernessLabs/Meadow_Issues #345: Cant do BLE Write Characteristics while WiFi is in use

jorgedevs, 2023-08-30:
Trying out the [ProjectLab Connectivity Sample](https://github.com/WildernessLabs/Meadow.ProjectLab.Samples/tree/main/Source/Connectivity), I noticed I can do Bluetooth read characterstics with the MAUI companion app (to get sensor data), but I cannot control the onboard LED (which is done by writing con BLE characteristics).

Noticed that this project has the WiFi configs to join the network, so the sample is activating Bluetooth while joining the network automatically. When I removed the auto join WiFi network, Bluetooth sample works as expected.

Workaround:

Removed the auto join WiFi feature and only join WiFi the C# way when using Maple, otherwise, Bluetooth will start only.

houstonhaynes, 2023-09-04:
Hey @jorgedevs I confirmed your issue with auto-join. I have my own project (solar tracker) where I'm doing some basic sensor checks and noted that I can't set the internal LED when WiFi is running with auto-join) 

Is there an F# way to join WiFi manually? 😸

houstonhaynes, 2023-09-04:
CORRECTION - I just restarted my Meadow with ` AutomaticallyStartNetwork: false` and see the same non-responsive behavior from my LED