WildernessLabs/Meadow_Issues #347: Sockets TcpListener and TcpClient "Lockstep"

E-R-T-W, 2023-09-07:
**Describe the bug**
Bug #1:
When using sockets, TcpListener and TcpClient, if there are 4 or more concurrent connections, and create a new Thread per connection, client connection exhibit a Lockstep bug.
Create a TcpListener on Feather F7v2 and a TcpClient in a PC Console App such that the Client sends the Server a message and the Server echos it back to the client.
All works well (almost, I'll describe a second bug below) if you have 3 or less connections.
As soon as you have 4 or more connections, round trip messages exhibit "lockstep" behaviour.

Bug #2:
Even with only a single client connection, if you close a StreamReader for the connection, the Thread freezes and is in "lockstep" until a second connection is established, at which point the first connection Thread unfreezes. 

**To Reproduce**
Bug #1:
Create 4 client connections and let's just consider connection 1 and connection 4...
Send a message from 4: the server (feather board) receives it, but it does not get echoed back to the client (PC console app).
Next send a message from 1: message 1 is sent round trip correctly, and unfreezes 4 so the client 4 receives the message back from the server.
Likewise...
Send a message from 1: the server (feather board) receives it, but it does not get echoed back to the client (PC console app).
Next send a message from 4: message 4 is sent round trip correctly, and unfreezes 1 so the client 1 receives the message back from the server.
Never is there any kind of an exception - everything still sort of works, it's just that instances/connections freeze, but only if 4 or more instances.

Bug #2:
Uncomment the reader.Close(); line in the Feather F7v2 Listener/Server App.
Create a Client Connection.
Set a Break Point on the next line.
Close the Client App or send the message "Exit" and code stops and freezes on the reader.Close() line, never reaching the Break Point.
Create a new Client Connection.
The first thread will unfreeze, and the Break Point will be hit, then the 2nd new connection will be established.

**Expected behavior**
Bug #1:
Regardless of how many Clients connect to the Server, messages should not exhibit "lockstep" behaviour i.e. in this example code, message should echo round trip from Client to Server and then back from Server to Client independent of other Clients having to send messages to the server.

Bug #2:
When a Client connection is closed, and reader.Close() is called, it should execute to completion and not be in "lockstep" with a new connection having to be made, to unfreeze the first Thread.

**Screenshots**
See Source Code for Meadow Feather F7v2 Listener/Server and Source Code for PC Console App Client, attached.

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 10 v22H2
 - IDE and version: Visual Studio Professional 2022 v17.7.3
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v1.3.0

**Meadow (please complete the following information as best as you can):**
 - Meadow hardware version: F7FeatherV2
 - Meadow OS version: 1.3.0.0
 - Meadow Mono version: 1.3.0.0
 - Meadow coprocessor version: 1.3.0.0

**Additional context**
This issue has already been discussed with @Alexis Christoforides on Slack #support on Monday, September 4th. Alexis asked me to create this Issue.
[Listener.zip](https://github.com/WildernessLabs/Meadow_Issues/files/12544470/Listener.zip)
[Client.zip](https://github.com/WildernessLabs/Meadow_Issues/files/12544472/Client.zip)

E-R-T-W, 2023-09-09:
I was doing some testing...
For Bug #2, of course if you attempt to use the work around of not closing the client, the reader, and the writer, then after 14 connections, it bombs with a too many files open (which makes sense).

alexischr, 2023-09-28:
There is some more useful context in the slack conversation: https://wildernesslabspublic.slack.com/archives/CQWD660E8/p1693864386772449

bryancostanich, 2023-10-17:
To share some more progress: Eduardo found that `ret = fcntl (((MonoFDHandle*) sockethandle)->fd, F_GETFL, 0);` is the line that’s blocking, in `w32socket-unix.c`

bryancostanich, 2023-10-19:
It's not closed until we ship.

alexischr, 2023-10-19:
Github  doing magic because I had the string "Fixes [issue_URL]...

duduita, 2023-10-23:
Just to share more progress related to this issue:
- The #bug 2 should be fixed in the Upcoming Release.
- The #bug 1 can be divided into two parts:
1. The maximum number of threads on the ESP32, responsible for the WiFi, was too low `4`, then it was increased to `8`, so now you'll be able to connect up to 7 clients (since the server consumes a thread as well).
2. When all the available threads are used and a new thread is attempted to be created, our current design on WiFi is to block the caller until a thread is released. Since it might lead to weird behaviors, like the "lockstep behavior", we decided to change our design to reject the threads, instead of blocking the caller, so I created a new [issue](https://github.com/WildernessLabs/Meadow_Issues/issues/372) for this implementation.

E-R-T-W, 2023-10-23:
Thank you so much!

It would good to include in the documentation the fact that 7 threads is the maximum allowed, thus 7 client connections maximum (plus the server connection).

Thanks again!

closed 2023-10-31 by bryancostanich

E-R-T-W, 2023-11-03:
Hi @bryancostanich, @duduita, @alexischr , @lamebrain, @NevynUK:

Thanks so much for working on this bug!

The Lockstep nature of this issue has indeed been resolved, but there's still a bug with the implementation of the fix. The Meadow Feather F7v2 will still accept 9 connections. Upon the 9th connection, after a "somewhat" successful connection, the connection gets closed (not by my C# code) and an error is reported stating:
`Not enough buffer space is available`
 At that point, no further connections are accepted, even after closing all of the other 8 connections. You are basically dead in the water at that point.
 
 The good news (a work around) is that now that the Lockstep bug is resolved, and I know how it is supposed to work, my C# code monitors the number of connections and the C# code uses the 7th connection to send a message back to the client explaining there are too many connections and closes the 7th connection. As long as the C# code manages to 6 active connections, and the 7th connection used to manage connection state, send a message back to the client, and then close the 7th connection, all is well.

I've attached the output of the problem.
[Output.txt](https://github.com/WildernessLabs/Meadow_Issues/files/13255361/Output.txt)

Do you need me to file a separate bug issue? Or can you open this issue #347 back up again?

Thanks for your help and hard work :)

duduita, 2023-11-03:
I believe that this behavior is expected in our current design, since what is happening is that after trying to connect more than the maximum available clients, the server thread is being killed to allow the 9th client.

duduita, 2023-11-04:
Besides that, just a caveat, when there's no thread available, the OS waits up to 5 seconds to see if a thread will be released, throwing an exception if there isn't. It explains why after a few seconds that you establish the 9th connection, you got the exception. You can have only 8 threads: 7 clients + 1 server.

E-R-T-W, 2023-11-04:
Understood! No worries, it works great! Thanks again!