WildernessLabs/Meadow_Issues #355: Network request tests

lamebrain, 2023-09-27:
I was able to create some network failures. Here are the results:

## Serial Requests
[App](https://gist.github.com/lamebrain/c7a69d3eebb9e0176dd03b600045392b) makes consecutive http requests

### without `using`
Consistently failed around 30 requests with the following errors:
```
Meadow StdOut: App Error in App Run: The descriptor is not a socket
Meadow StdOut: System.Net.Http.HttpRequestException: The descriptor is not a socket
Meadow StdOut: at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) [0x001ac] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at System.Net.Http.HttpConnectionPool.CreateConnectionAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00134] in <1ea076ff21e0452
Meadow StdOut: bf7dd682df3dafc6>:0
Meadow StdOut: at System.Net.Http.HttpConnectionPool.WaitForCreatedConnectionAsync (System.Threading.Tasks.ValueTask`1[TResult] creationTask) [0x000a2] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at System.Net.Http.HttpConnectionPool.SendWithRetryAsync (System.Net.Http.HttpRequestMessage request, System.Boolean doRequestAuth, System.Threading.CancellationToken cancellationT
Meadow StdOut: ken) [0x00089] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at System.Net.Http.RedirectHandler.SendAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00094] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at System.Net.Http.HttpClient.FinishSendAsyncBuffered (System.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequ
Meadow StdOut: stMessage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) [0x0016c] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at MeadowApp.MeadowApp.Run () [0x000d9] in <d8a391ac22de4389b1dfde035b7d2bce>:0
Meadow StdOut: at Meadow.MeadowOS.Start (System.String[] args, Meadow.IApp app) [0x00164] in <d4f614572c6047c08610bc5349864be2>:0
Meadow StdOut: Inner System.Net.Sockets.SocketExcepti
Meadow StdOut: n: The descriptor is not a socket
Meadow StdOut: at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) [0x00110] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: App shutting down
Meadow StdOut: Shutdown
```
and 
```
Meadow StdOut: mono_fdhandle_insert: duplicate Socket fd 39
Meadow StdOut: mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
Meadow StdOut: code '142'" (142)
```
### with `using`

This works reliably as I got over 200 consecutive successful results.

## Parallel Requests
[App](https://gist.github.com/lamebrain/07d1ab53d29e81ded4cdb0fbc7850048) makes a new request every 500ms

### without `using`
Consistently failed around 20 requests
```
Meadow StdOut: System.Net.Http.HttpRequestException: Too many open files
Meadow StdOut: at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) [0x001ac] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at System.Net.Http.HttpConnectionPool.CreateConnectionAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00134] in <1ea076ff21e0452
Meadow StdOut: bf7dd682df3dafc6>:0
Meadow StdOut: at System.Net.Http.HttpConnectionPool.WaitForCreatedConnectionAsync (System.Threading.Tasks.ValueTask`1[TResult] creationTask) [0x000a2] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at System.Net.Http.HttpConnectionPool.SendWithRetryAsync (System.Net.Http.HttpRequestMessage request, System.Boolean doRequestAuth, System.Threading.CancellationToken cancellationT
Meadow StdOut: ken) [0x00089] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at System.Net.Http.RedirectHandler.SendAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00094] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at System.Net.Http.HttpClient.FinishSendAsyncBuffered (System.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequ
Meadow StdOut: stMessage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) [0x0016c] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at MeadowApp.MeadowApp.DoHttpRequest (System.Int32 instance) [0x000c1] in <17b470bd590e4e1ca864cf9aa72f2f39>:0
Meadow StdOut: at MeadowApp.MeadowApp+<>c__DisplayClass0_0.<Run>b__0 (System.Object sender, System.Timers.ElapsedEventArgs args) [0x00089]
Meadow StdOut: n <17b470bd590e4e1ca864cf9aa72f2f39>:0
Meadow StdOut: at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__7_1 (System.Object state) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading.QueueUserWorkItemCallback.WaitCallbac
Meadow StdOut: k_Context (System.Object state) [0x00007] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading.ExecutionContext.Run (Syste
Meadow StdOut: .Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading.ThreadPoolWor
Meadow StdOut: Queue.Dispatch () [0x00074] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: Inner System.Net.Sockets.SocketException: Too many open files
Meadow StdOut: a
Meadow StdOut: t System.Net.Sockets.Socket..ctor (System.Net.Sockets.AddressFamily addressFamily, System.Net.Sockets.SocketType socketType, System.Net.Sockets.ProtocolType protocolType) [0x00069] in <b25afc10f7a8494fb2c79721c1c70004>:0
Meadow StdOut: at System.Net.Sockets.DualSocketMultipleConnectAsync..ctor (System.Net.Sockets.SocketType socketType, System.Net.Sockets.ProtocolType protocolType) [0x0000d] i
[... 19 lines cut ...]
Meadow StdOut: at System.Net.Http.HttpClient.FinishSendAsyncBuffered (System.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequ
Meadow StdOut: stMessage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) [0x0016c] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: at MeadowApp.MeadowApp.DoHttpRequest (System.Int32 instance) [0x000c6] in <7710f2330542423589fff7695ec79798>:0
Meadow StdOut: at MeadowApp.MeadowApp+<>c__DisplayClass0_0.<Run>b__0 (System.Object sender, System.Timers.ElapsedEventArgs args) [0x00089]
Meadow StdOut: n <7710f2330542423589fff7695ec79798>:0
Meadow StdOut: at System.Runtime.CompilerServices.AsyncMethodBuilderCore+<>c.<ThrowAsync>b__7_1 (System.Object state) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading.QueueUserWorkItemCallback.WaitCallbac
Meadow StdOut: k_Context (System.Object state) [0x00007] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading.ExecutionContext.RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading.ExecutionContext.Run (Syste
Meadow StdOut: .Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading.QueueUserWorkItemCallback.System.Threading.IThreadPoolWorkItem.ExecuteWorkItem () [0x00021] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading.ThreadPoolWor
Meadow StdOut: Queue.Dispatch () [0x00074] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
Meadow StdOut: at System.Threading._ThreadPoolWaitCallback.PerformWaitCallback () [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0
[... cut ...]
Meadow StdOut: in <b25afc10f7a8494fb2c79721c1c70004>:0
Meadow StdOut: at System.Net.Sockets.Socket.ConnectAsync (System.Net.Sockets.SocketType socketType, System.Net.Sockets.ProtocolType protocolType, System.Net.Sockets.SocketAsyncEventArgs e) [0x00056] in <b25afc10f7a8494fb2c79721c1c70004>:0
Meadow StdOut: at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.Cancellat
Meadow StdOut: onToken cancellationToken) [0x00058] in <1ea076ff21e0452abf7dd682df3dafc6>:0
Meadow StdOut: Shutdown
```

DarkNoir, 2023-12-15:
This happens when using 3 tasks, each sending data using HTTPS POST at different intervals (5,6 and 7 seconds) on their own instance of HttpClient that is only used for a single call then disposed:
```
mono_fdhandle_insert: duplicate Socket fd 33
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
```

duduita, 2023-12-15:
Just to confirm, are you facing this issue even with the `using` keyword and/or `GC.collect()`? @DarkNoir 

I'm working on the duplicated socket issue error, but I couldn't reproduce it using these workarounds.

DarkNoir, 2023-12-15:
yes, I am using 'using' the project i shared with you on Github is currently updated to reproduce this error:
https://github.com/DarkNoir/MeadowHttpClientErrors.git

duduita, 2023-12-18:
Let me summarize what I could conclude for now:

- Serial Requests without `using` :
Sometimes a socket descriptor is closed by the OS, but doesn't notify Mono, which will only realize that the socket is not available after a garbage collector cycle, letting them out of sync in this meantime, causing the duplicate file descriptor error. I still trying to understand why the OS is closing sometimes the sockets without Mono asking for it.

- Serial Requests with `using` :
Anyway, the `using` keyword makes the sockets with no reference close right after a request, so we don't need to wait for the GC cycle. **Then, `using` seems a nice workaround for the duplicate file descriptor issue.** This works reliably as I got over 25k consecutive successful requests without getting the duplicate file descriptor error @alexisch.

- Parallel Requests without using:
The OS has a limited number of file descriptors, so when this limit is achieved after many parallel requests, Mono throws the exception `Too many open files`, which makes sense, and it's expected in my opinion.

- Parallel Requests with using:
It requires more investigation, anyway, using Ethernet I got a different error:
```
Meadow StdOut: mono_w32socket_convert_error: no translation into winsock error
Meadow StdOut: for (16) "Device or resource busy"
```

bryancostanich, 2024-01-04:
Looks like we've got a fix for this, it's getting final polish and then we'll get it tested internally.

alexischr, 2024-01-05:
This Github Issue describes two separate defects both manifesting when the network is put under pressure: one marked with `duplicate Socket fd" printed by the runtime to the console, and one marked with "The descriptor is not a socket" as the text in a thrown SocketException. @duduita has a candidate fix for the one in the title - "duplicate Socked fd". I suggest we close this issue when the patch is merged, and we move discussion of the second bug to this issue: https://github.com/WildernessLabs/Meadow_Issues/issues/380

bryancostanich, 2024-01-09:
We might have a fix! It takes hours to validate, however. Hoping to get this into the v1.8 release.

alexischr, 2024-01-16:
Fixed in 1.8

closed 2024-01-16 by alexischr