WildernessLabs/Meadow_Issues #358: Former, rarely used QSPI memory test command removed from CLI v2. Need set developer test added.

Peter-Moody, 2023-10-02:
**Describe the bug**
The v1 CLI has 3 commands to test QSPI external flash memory. These were created for testing and developing the driver for the FeatherV2 Meadow board. The v2 of CLI doesn't (and shouldn't) support these 3 commands. However, it would be best to add a `set developer` test to exercise these 3 commands.
The 3 commands are `qspi init`, `qspi read` and `qspi write`.

CartBlanche, 2023-10-02:
I believe we already have a set developer command. @ctacke ?

ctacke, 2023-10-02:
CLI does, but there are no command numbers for QSPI operations in the underlying HCOM.  That's what Peter is suggesting to be added.

Peter-Moody, 2023-10-02:
Yes, in CLI (both versions) there is the `set developer` command. But, as Chris stated, we need to implement the Meadow end in the OS to recognize the command and route it to the, now orphaned, QSPI code that executes the tests.

Peter-Moody, 2025-02-07:
This test may never be used again. So, as long as the code remains in Meadow OS code base, it can be made available if it is ever needed. The code is never built unless it is needed.

closed 2025-02-07 by Peter-Moody