WildernessLabs/Meadow_Issues #359: Project Lab button events don't fire after reading DateTime.Now

adrianstevens, 2023-10-03:
Project Lab v3.x
Meadow OS 1.3.4.0 & 1.3.1.8 using local project references

If I access DateTime.Now before creating the project lab instance, button events aren't raised (no error). If I access DateTime.Now after it works as expected. This is true for all 4 buttons on the device.

I couldn't reproduce this behavior on Project Lab v1.g

```
using Meadow;
using Meadow.Devices;
using System;
using System.Threading.Tasks;

namespace EventRepro
{
    // Change F7FeatherV2 to F7FeatherV1 for V1.x boards
    public class MeadowApp : App<F7CoreComputeV2>
    {
        IProjectLabHardware projLab;

        public override Task Initialize()
        {
            Console.WriteLine("Initialize...");

            Console.WriteLine($"Now {DateTime.Now}"); //clicked event not raised
            // Console.WriteLine($"Now {DateTime.UtcNow}"); //clicked event raised
            // Console.WriteLine($"Now {DateTimeOffset.Now}"); //clicked event not raised
            // Console.WriteLine($"Now {DateTimeOffset.UtcNow}"); //clicked event not raised

            projLab = ProjectLab.Create();

            //Console.WriteLine($"Now {DateTime.Now}"); //clicked event raised
            projLab.DownButton.Clicked += (s, e) => Console.WriteLine("Down clicked");

            Console.WriteLine("Init complete");
            return Task.CompletedTask;
        }

        public override Task Run()
        {
            Resolver.Log.Info("Run ...");
            return Task.CompletedTask;
        }
    }
}
```

alexischr, 2023-10-10:
Switching to Pri-O and assigning @halyssonJr  to this - this is a nasty one. From the user's perspective, they won't think  to check if they "used the clock" or not, and how they did that.

@adrianstevens I seem to remember you saying that DateTime.UtcNow usage does not break button events, only DateTime.Now, is that correct?

adrianstevens, 2023-10-16:
@alexischr yes exactly - you can see what we tested and what worked in the code snippet above

```
Console.WriteLine($"Now {DateTime.Now}"); //clicked event not raised                                 <- fails
// Console.WriteLine($"Now {DateTime.UtcNow}"); //clicked event raised                              <- works
// Console.WriteLine($"Now {DateTimeOffset.Now}"); //clicked event not raised                    <- fails
// Console.WriteLine($"Now {DateTimeOffset.UtcNow}"); //clicked event not raised               <- fails
```

alexischr, 2023-10-17:
@adrianstevens @halyssonJr do you think we can work on a different reproduction that does not require the ProjectLab HW? Halysson's is not functioning properly, and it's possible that this just requires one single button hooked on the Meadow.

adrianstevens, 2023-10-26:
And to add more info - this PR also breaks button events on ProjectLab V3
https://github.com/WildernessLabs/Meadow.Core/commit/f4cc7208fbe8c920a8063aadcd508326add6dadb

Which is the opposite problem (works w DateTime.Now, fails w DateTime.UtcNow)

adrianstevens, 2023-10-26:
I haven't been able to reproduce this behavior on a Feather F7 V2 connected to either Project Lab v1.g or v2.e.

bryancostanich, 2023-10-31:
Likely related to #311 according to the latest investigations

bryancostanich, 2023-11-02:
After more investigation, it looks like `ProjectLab.Create()` doesn't return unless you press a button. We're still digging.

bryancostanich, 2023-11-07:
Looks like this is fixed with the Round-Robin #311 fixes.

alexischr, 2023-11-07:
Let's wait until https://github.com/WildernessLabs/Meadow/pull/502 is reasoned with, and merged @bryancostanich

adrianstevens, 2024-03-01:
We still have a DateTime.Now issue with interrupt that apears to be a race condition

adrianstevens, 2024-03-01:
In DigitalInterruptPort.cs in Meadow.Core, in the OnInterrupt method, if the LastEventTime assignment is changed to DateTime.UtcNow, events don't fire correctly on the ProjectLab v3 buttons.

If we add a `Thread.Sleep(100)` before the assignment .... events fire again.

adrianstevens, 2024-07-24:
Updated link https://github.com/WildernessLabs/Meadow.Core/blob/develop/Source/Meadow.Core/Hardware/DigitalInterruptPort.cs#L138

adrianstevens, 2024-07-24:
Can't repro with v1.13.0.0 .... I think we can mark as fixed

closed 2024-09-10 by bryancostanich

alexischr, 2024-09-23:
For posterity: this was fixed by https://github.com/WildernessLabs/Meadow/pull/630