WildernessLabs/Meadow_Issues #360: Meadow.OS will not startup if a "bad" meadow.config.yaml configuration file has been downloaded

Peter-Moody, 2023-10-03:
**Describe the bug**
There have been instances where a `meadow.config.yaml` file loaded into Meadow will cause the OS to fail to completely start. However, this reveals a larger problem, there are likely other situation that can prevent the `Meadow.OS` from successfully starting.

**To Reproduce**
Steps to reproduce the behavior:
1. Download a bad configuration file. What it is about the `meadow.config.yam`l file is currently not known.

Peter-Moody, 2023-10-03:
On 3Oct23 there was a slack discussion below is a summary of the ideas proposed.
1. If boot fails on next boot enter some kind of "safe" mode
 - Pro-Simpler to implement and would allow those with bad config files etc. to recover.
 - Con-Requires user to resolve the problem. Assumes `Nuttx `and the `HCOM `can communicate with CLI.
2. Monitor startup progress to help diagnose at what stage the system crashed
 - Pro-Simple to implement and provides diagnostic information.
 - Con-Not intended to solve the problem, but to augment the diagnostic effort.
3. Flash a `Primary `and `Secondary `version of the OS. If `Primary `version fails to boot switch to the `Secondary `version. The `Secondary `OS can be preinstalled in internal flash or in the file system.
 - Pro-Provides a `Secondary `OS that can be fully tested to insure success.
 - Con-More complex to implement and there are likely unknow issues to resolve.
4. Do something utilizing the `bootloader`
 - Pro-Bootloader is already part of the system, no new components. It has full access to system and could automatically load a working OS version and then reboot the system automatically.
 - Con-Would not solve the current bad configuration file issue. Currently, no well defined functionality for what the bootloader would do in various situations.

Peter-Moody, 2023-10-04:
Reduced priority to 1 as it seems there really isn't a critical need for this.

doingnz, 2023-10-04:
Re the bootloader and configuration file issue, it might be possible to check in the bootloader if the BOOT button is pressed for an "extended" period of time after reset is released. This could be taken as a request to erase all configuration files, assuming the Meadow.OS would boot correctly with no configuration files.

Peter-Moody, 2023-10-05:
Checking the BOOT button after startup is possible on _some_ STM32F MCU's but not the one we're using (STM32F777iik6). I checked the documents 2 days ago because a team member suggested something much like you did.
Thanks and keep the ideas coming!

NevynUK, 2023-11-18:
Version 0.0.7.0 has been created to facilitate local recovery.

halyssonJr, 2025-01-02:
This issue was fixed. If we download a file with an unexpected (e.g., bad indentation), the OS/Core uses the default configuration and sends a message informing us that the file is invalid, but no crash.

```
[esp] Processing event.
[esp] Simple event, interface 4, event code: 12, status code 0x0000002b
[esp] Event data collected, raising event.
[esp] Waiting for event.
[core] The ESP32 has had an error (InvalidConfigurationFile).
[core] PlatformOS Initialize starting...
[core] SD Card is configured to 'off'
[core] File system Initialize starting...
[core] Creating instance of MeadowApp...
[core] Creating 'MeadowApp' instance took 68ms
...
```

halyssonJr, 2025-01-02:
@ctacke  and @alexischr, should we close this issue?

closed 2025-01-03 by alexischr