WildernessLabs/Meadow_Issues #361: HCOM stops responding to CLI after some time has passed

alexischr, 2023-10-03:
**Describe the bug**
`meadow listen` or other CLI commands that connect to Meadow.OS's HCOM, stop working if the Meadow has been running for some time and is not already connected to with `meadow listen`.

**To Reproduce**

((@halyssonJr please edit and fill in with repro details))
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
`meadow listen` and HCOM CLI commands should always work, as long as the Meadow has not crashed.

**Meadow (please complete the following information as best as you can):**
* All Meadow hardware is affected
* All Meadow OS versions are affected

ctacke, 2023-10-05:
Repro:

- Create or modify a simple Managed app
- Set the platform to the wrong thing (i.e. set public class MeadowApp : App<F7FeatherV1> but deploy to a CoreCompute device
- Modify (or add) app.config.yaml to have the following:
```
Lifecycle:
  RestartOnAppFailure: false
```

The device will boot then give something like this:

```
Meadow StdOut: Application is defined as F7FeatherV2, but running hardware is F7FeatherV1
Meadow StdOut: Device (system) Initialization Failure
```

At this point HCOM is non-responsive. Even meadow device info will timeout.

bryancostanich, 2024-01-09:
We haven't seen this in a while, but we need to validate using @ctacke's repro above.

bryancostanich, 2024-01-15:
Thanks! I wonder if it got fixed before this release. In any case, since you validated in `v1.7.1.2`, i'll mark as fixed in upcoming release.

closed 2024-03-22 by alexischr