WildernessLabs/Meadow_Issues #363: OS OtA not working

lamebrain, 2023-10-06:
I created an OtA package with a OS version of 1.3.4.0, board was running 1.3.0.0.
OtA process downloaded and copied fine, but after the device restarted, the board was unresponsive.
After one manual reset, it remained unresponsive, ie could not connect to device via meadow listen.
After another manual reset, the board came alive with the original os version (1.3.0.0) and app. seems like the rollback also gets stuck.

[purple_134_withos.zip](https://github.com/WildernessLabs/Meadow_Issues/files/12833815/purple_134_withos.zip)

bryancostanich, 2023-10-17:
Update, this is two issues:
 * Bootloader fails validation on binary -  in progress
 * Doesn't reset after flashing userspace partition boot doesn't work - forcing a reset from code now.

bryancostanich, 2023-11-03:
Validated by @lamebrain. Will go out in v1.5

closed 2023-11-30 by bryancostanich