WildernessLabs/Meadow_Issues #365: Seems meadow file delete command accesses the file system while Mono is enabled

halyssonJr, 2023-10-10:
**Describe the bug**
For simple apps (blinky) if I use **`meadow file delete`** all while mono is enabled, the **ProjLab v3.a**  stop works. Seems to **`meadow file delete all`**  allow Meadow file access while mono is enabled. But, when you have a more complex app (CPU busy) and try to use this command even doesn't execute, but keeps the terminal open and works like **`meadow listen`** and I still received some confirmation from CLI that the file started to delete.

**To Reproduce**
Steps to reproduce the behavior:
1. Create or Modify an App
2. Build and Deploy
3. Check if the app is running (**`meadow listen`**)
4. Try using **`meadow file delete all`**

**Expected behavior**
Mono must be disabled for Meadow file system access.

**Screenshots**
- Case 1

[image]

- Case 2

[image]

**Developer tools:**
 - OS and version: Ubuntu 22.04.3 LTS x86_64
 - IDE and version: Version: 1.83.0
 - Meadow extension for IDE version: VS 1.3.4

**Meadow:**
 - Meadow hardware version: F7CoreComputeV2
 - Meadow OS version: v1.3.4.0

Peter-Moody, 2023-12-25:
@halyssonJr Can you retest this once after the next Meadow.OS release (1.7 probably).

adrianstevens, 2024-03-26:
Meadow.CLI v2 will disable the runtime before deleting

closed 2024-04-11 by halyssonJr