WildernessLabs/Meadow_Issues #366: Meadow.Cloud `ICommand` classes won't load if they're in a referenced library; they must be in the executing assembly.

bryancostanich, 2023-10-12:
**Describe the bug**

If you put your `ICommand` concrete type class in a library, say `Cultivar.Core` project, then reference them in say `Cultivar.MeadowApp` executable, you'll get errors. I can't remember the error specifically. But If you move those classes into the main Meadow App project they work fine.

**To Reproduce**
Steps to reproduce the behavior:
1. Open the Cultivar solution [here](https://github.com/WildernessLabs/Cultivar/tree/main/source/apps/Cultivar).
2. Move the commands into the `Cultivar.Core` library.
3. Run `MeadowApp` on a Project Lab and send a command using the cloud, targeting one of those classes.

**Expected behavior**
Command would come through.

**Developer tools (please complete the following information as best as you can):**
 - OS and version: n/a
 - IDE and version: n/a
 - Meadow extension for IDE version: n/a

**Meadow (please complete the following information as best as you can):**
Project Lab

**Additional context**
@lamebrain has additional context. I'm in transit.

ctacke, 2023-10-12:
The problem is not unique to this project.  Any items that get loaded via reflection are at risk of being removed by the linker during the trimming operation.  We do have the ability to mark an entire assembly to not be linked.

- Add a file named `app.build.yaml` to your project and ensure it gets copied to the output folder.
- add the following contents (this is an example, so adjust the assembly names for your project)
```
Deploy:
  nolink: {App.dll, MyCommandAssembly.dll}
  IncludePDBs: false
```

stevenkuhn, 2023-10-17:
I was able to get it working as @ctacke said by having file named `app.build.yaml` that is always copied to the output folder. The contents shoudl be:

```yaml
Deploy:
  NoLink: [ App, MyCommandAssembly ]
```

There will be a future update that will strip out `.dll` if it's included in the assembly name.

closed 2023-10-17 by stevenkuhn