WildernessLabs/Meadow_Issues #371: static HttpClient is really, really slow

lamebrain, 2023-10-21:
Serial static `HttpClient` requests are abnormally slow (and suspiciously consistent) :

```
Meadow StdOut: start 545
Meadow StdOut: end 545, success:True, elapsed: 00:01:02.8780000
Meadow StdOut: start 546
Meadow StdOut: end 546, success:True, elapsed: 00:01:02.4580000
Meadow StdOut: start 547
Meadow StdOut: end 547, success:True, elapsed: 00:01:02.3880000
Meadow StdOut: start 548
Meadow StdOut: end 548, success:True, elapsed: 00:01:02.3380000
Meadow StdOut: start 549
Meadow StdOut: end 549, success:True, elapsed: 00:01:02.3580000
Meadow StdOut: start 550
Meadow StdOut: end 550, success:True, elapsed: 00:01:02.0580000
Meadow StdOut: start 551
Meadow StdOut: end 551, success:True, elapsed: 00:01:02.3680000
Meadow StdOut: start 552
Meadow StdOut: end 552, success:True, elapsed: 00:01:02.6580000
Meadow StdOut: start 553
```

Repro: https://gist.github.com/lamebrain/4027138a1e3008d1897ea131bf1e4546

adrianstevens, 2023-10-22:
Would love to see this addressed soon

DarkNoir, 2023-10-26:
I have observed this also, had to switch to disposing of it after use (Sending data every minute)

adrianstevens, 2023-10-26:
The Meadow TestSuite validation app shows similar behavior

```
Running WiFiSSLLoopTest`1...
Connecting to valid SSID with valid passcode...
Sending command to connect
Sending ESP command. destination:1, function:2
ESP Ioctl returned success: WiFiAlreadyStarted
SendingCommand returned: WiFiAlreadyStarted
Failed to connect: The operation has timed out.
Requesting https://postman-echo.com/get?foo1=bar1&foo2=bar2 - 10/26/2023 03:24:02 #0
{
"args": {
"foo1": "bar1",
"foo2": "bar2"
},
"headers": {
"x-forwarded-proto": "https",
"x-forwarded-port": "443",
"host": "postman-echo.com",
"x-amzn-trace-id": "Root=1-653a99fd-4efa0919360ad1a275aeb2eb"
},
"url": "https://postman-echo.com/get?foo1=bar1&foo2=bar2"
}
```

`Failed to connect: The operation has timed out.` delays about 60s

bryancostanich, 2023-11-14:
Sounds like the TLS section is slowing things down, specifically the Random Number Generator (RNG) is slow right now for some reason.

bryancostanich, 2023-11-21:
More progress, this isn't the RNG, but something to do with the ESP32 integration. Digging in there now.

bryancostanich, 2023-11-30:
Some additional progress here. From @halyssonJr:

In the _mbedtls_ `net recv` function, we read a file descriptor which is precisely where most of the execution is spent, that is, it is blocked.

DarkNoir, 2023-12-04:
Keep an eye out for the memory leak while you are poking around in there

bryancostanich, 2024-01-30:
Possibly fixed in v1.8. @halyssonJr to dig in.

bryancostanich, 2024-02-15:
We have a workaround for this. @halyssonJr to post. I'm lowering to p-1.

duduita, 2024-04-01:
I realized that this delay is caused by the HTTP keep-alive stuff.

A way to work around it is by adding a header `Connection: "close"` to your `DoHttpRequest` method @lamebrain:
```
async Task DoHttpRequest(int instance)
{
    Console.WriteLine($"start {instance}");
    string payload = "hello";

    sw.Start();

    using (var content = new StringContent(payload, Encoding.UTF8))
    {
        _httpClient.DefaultRequestHeaders.Connection.Add("close");
        using (var result =
            await _httpClient.PostAsync("post", content).ConfigureAwait(false))
        {
            sw.Stop();
            Console.WriteLine($"end {instance}, success:{result.IsSuccessStatusCode}, elapsed: {sw.Elapsed.ToString()}");
            sw.Reset();
        }
    }
}
```

closed 2024-09-22 by duduita