WildernessLabs/Meadow_Issues #374: Add timestamp to report the OS tick count for each interrupt

Peter-Moody, 2023-10-27:
**Is your feature request related to a problem? Please describe.**
Currently, Meadow relies on the managed code to timestamp events/interrupts that occur in the system. This  needs to be changed so that the OS adds a timestamp as close to the time the event occurred as possible and pass this information to the managed side for final processing.

NevynUK, 2023-10-27:
I'd like to suggest that we use a time unit rather than ticks.  This could be microseconds, milliseconds or something in between.

Justification: If we change the clock speed on the current platform or move to another platform with different clock speeds then any user applications that use a time based calculation will need modifying.  Without the unit all the timestamp allows is the determination of the order in which events happen.

Peter-Moody, 2023-11-10:
Look at issue #346 at the same time.

closed 2025-12-04 by Peter-Moody