WildernessLabs/Meadow_Issues #375: GPIO State Transitions on Power-up/Start-up/Reset

E-R-T-W, 2023-10-28:
### **Describe the bug**
On Power-up/Start-up/Reset, GPIO changes state, **prior to configuring GPIO and taking control of the GPIO in C# code**. In many scenarios this is problematic and unacceptable.

### **To Reproduce**
For example, let's discuss driving a PNP transistor switch as shown below with the Base connected to GPIO. Anytime the GPIO sinks current, the transistor will switch ON.
[image]

Via repeated testing, I've attached the "definitive" GPIO state transitions during Power-up/Start-up/Reset:
[image]

Some explanation:
There are really only 2 GPIO **cases** for each scenario to be discussed:
1. GPIO A00-A02, A05, D05-D08, D14-D15
2. GPIO A03-A04, D00-D04, D09-D13

Prior to even discussing the 3 Scenarios once the C# code is up and running, all GPIO for case #1 remain high impedance for ~19 seconds (~14 seconds while blue LED ON and an additional ~5 seconds while blue LED OFF), but all GPIO for case #1 are in a high impedance state for ~14 seconds (while blue LED ON), followed by sinking current for ~5 seconds (*while blue LED OFF).

At this point I believe is when the C# code executes. I've examined 3 **scenarios**:
1. Configure all GPIO to Inputs
2. Configure all GPIO to Output false
3. Configure all GPIO to Output true

**Scenario #1 Case #1 GPIO:**
These GPIO are high impedance for ~14 seconds while blue LED ON, then continue high impedance for ~5 seconds while blue LED OFF.
Next the C# code executes; GPIO are configured to Input, so all GPIO are high impedance. 
There's no problem with Secnario #1 Case #1 GPIO.

**Scenario #1 Case #2 GPIO:**
These GPIO are high impedance for ~14 seconds while blue LED ON, then start to sink current for ~5 seconds while blue LED OFF.
Next the C# code executes; GPIO are configured to Input, so all GPIO are high impedance. 
This is a problem with the change in GPIO state for Case #2 GPIO in this Scenario #1.

**Scenario #2 Case #1 GPIO:**
These GPIO are high impedance for ~14 seconds while blue LED ON, then continue high impedance for ~5 seconds while blue LED OFF.
Next the C# code executes; GPIO are configured to Output false, so all GPIO sink current. 
There's no problem with Secnario #2 Case #1 GPIO.

**Scenario #2 Case #2 GPIO:**
These GPIO are high impedance for ~14 seconds while blue LED ON, then start to sink current for ~5 seconds while blue LED OFF.
Next the C# code executes; GPIO are configured to Output false, so all GPIO sink current.
This is a problem with the change in GPIO state for Case #2 GPIO in this Scenario #2.

**Scenario #3 Case #1 GPIO:**
These GPIO are high impedance for ~14 seconds while blue LED ON, then continue high impedance for ~5 seconds while blue LED OFF.
Next the C# code executes; GPIO are configured to Output true, so all GPIO go high and should not sink current, however, what happens is that when one first configures the GPIO to Outputs, they sink current for a fraction of a second, until the code sets the Outputs to true.
This "blip" of a change in state to sinking current until the Case #1 GPIO are set to true is a problem in this Scenario #3.

**Scenario #3 Case #2 GPIO:**
These GPIO are high impedance for ~14 seconds while blue LED ON, then start to sink current for ~5 seconds while blue LED OFF.
Next the C# code executes; GPIO are configured to Output true, so all GPIO go high. There is no "blip" of change in state like Scenario #3 Case #1 because there was already a ~5 second change of state to sinking current while blue LED OFF.
This is a problem with the change in GPIO state for Case #2 GPIO in this Scenario #3. Case #2 GPIO should not sink current for ~5 seconds in this Scenario #3.

**Code for Scenario #1:**
```
        IDigitalInputPort inputA00;
        IDigitalInputPort inputA01;
        IDigitalInputPort inputA02;
        IDigitalInputPort inputA03;
        IDigitalInputPort inputA04;
        IDigitalInputPort inputA05;
        IDigitalInputPort inputD00;
        IDigitalInputPort inputD01;
        IDigitalInputPort inputD02;
        IDigitalInputPort inputD03;
        IDigitalInputPort inputD04;
        IDigitalInputPort inputD05;
        IDigitalInputPort inputD06;
        IDigitalInputPort inputD07;
        IDigitalInputPort inputD08;
        IDigitalInputPort inputD09;
[... 207 lines cut ...]
        }
```

### **Expected behaviour**
GPIO state should not change during the Start-up/Power-up/Reset phase.

### **Screenshots**
[image]

### **Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11 Pro v22H2
 - IDE and version: Visual Studio Professional 2022 v17.7.5
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v1.3.4

### **Meadow (please complete the following information as best as you can):**
Most of these values can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7v2
 - Meadow OS version: 1.3.4.0
 - If they are different, please provide these versions as well.
   - Meadow Mono version: 1.3.4.0
   - Meadow co-processor version: 1.3.4.0

### **Additional context**
Further details and information can be found on Slack in a discussion with chris.tacke@wildernesslabs.co (he asked me to create this issue) at:
https://wildernesslabspublic.slack.com/archives/CQWD660E8/p1698019417572109.

ctacke, 2023-11-01:
Release 1.4 changes how Meadow.Core initializes GPIOs.  They are now set as inputs with no resistor, so should effectively become high-z.  For all of the pons in the table above where it's "OFF OFF OFF" these should behave as expected.  We're still investigating the others

E-R-T-W, 2023-11-05:
I've had a chance to tested - looking good - thanks for the progress!!

Just one last issue (as you already know). I've updated the table as follows:
[image]

There is only one issue left for all GPIO (they now all consistently behave the same):
When configure all GPIO to **output true**, there's a short "blip" for a few milliSeconds where the GPIO sink current, between the time the GPIO are configured to outputs, and when they are set to true.

All other cases/scenarios have been resolved - thanks again!!

E-R-T-W, 2023-11-05:
By the way, the blue RGB LED has this same short "blip" upon Power-up/Start-up/Reset (after it goes through it's regular ~14 seconds blue LED ON followed by ~5 seconds blue LED OFF prior to the C# code executing). I assume it will be resolved too, along with all other GPIO that have the same issue...

alexischr, 2025-01-16:
@E-R-T-W Instead of:
```
outputA00 = Device.CreateDigitalOutputPort(Device.Pins.A00);
```

please use the alternative constructor 
```
outputA00 = Device.CreateDigitalOutputPort(Device.Pins.A00. true);
```

As the default state for the digital output port is false.

closed 2025-01-16 by alexischr