WildernessLabs/Meadow_Issues #376: Don't deploy PDB and other files not needed to run Meadow application

doingnz, 2023-10-29:
**Describe the bug**
Visual Studio tools deploy .pdb files to Meadow when deploying assemblies for an application. They are not required for debugging Meadow application with Visual Studio as it uses the pdb files on the PC. These pdb files unnecessarily consume space in the onboard flash storage and are typically large and add to the time it takes to deploy the application. When OTA deployments happen, the pdb files will not match the deployed new dlls.

**To Reproduce**
Steps to reproduce the behavior:
1. Rebuild a Meadow application and deploy from Visual Studio.  
2. Monitor the files deployed from the PC and look for PDB files in the list. Typically included for the App.dll and any other locally built assemblies. 

**Expected behavior**
Debugger files with `.pdb` extension should not be deployed to Meadow.

**Screenshots**
Example output from `Meadow` window in Visual Studio during deployment of an application.
Note in addition to `App.pdb`, other unused files can also be deployed. e.g. `App.deps.json`, ` Bme68x.xml`, etc.
```
 Download of 'App.dll' success (checksums calculated:0x7473855A, expected:0x7473855A)
 File 73% downloaded
 Download of 'App.pdb' success (checksums calculated:0xA411AACA, expected:0xA411AACA)
 File 81% downloaded
 Download of 'App.deps.json' success (checksums calculated:0xFF0491A7, expected:0xFF0491A7)
 File 67% downloaded
 Download of 'Bme68x.xml' success (checksums calculated:0x87EAD72F, expected:0x87EAD72F)
 File 88% downloaded
 Download of 'Mcp23xxx.xml' success (checksums calculated:0xAF07D6A8, expected:0xAF07D6A8)
 File 91% downloaded
```

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Microsoft Windows [Version 10.0.22621.2506]
 - IDE and version: Microsoft Visual Studio Enterprise 2022 (2) (64-bit) - Current Version 17.7.5
 - Meadow extension for IDE version: 1.3.4.0

**Meadow (please complete the following information as best as you can):**
```
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7CoreComputeV2
    Device name: MeadowCompute_Clima

Hardware Information
    Processor type: STM32F777IIK6
    ID: 34-00-25-00-03-51-38-34-35-31-38-33
    Serial number: 335D31733438
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 4C:75:25:D7:13:C0

Firmware Versions
    OS: 1.3.4.0
    Mono: 1.3.4.0
    Coprocessor: 1.3.4.0
    Protocol: 7
```

**Additional context**
To test debugging works with the PDB files deleted, add the following to list all files and erase any `*.pdb` files. Call  from `Initialize()` as `DirList("/");` . You can add and remove application breakpoints after the PDF files are deleted vai the Visual Studio UI and can single step the Meadow application, view variables, etc.

```
        static void DirList(string sDir)
        {
            try
            {
                foreach (string d in Directory.GetDirectories(sDir))
                {
                    foreach (string f in Directory.GetFiles(d))
                    {
                        Resolver.Log.Trace(f);
                        if (Path.GetExtension(f).ToLower().Contains("pdb"))
                        {
                            Resolver.Log.Trace($"Delete PDB: {f}");
                            File.Delete(f);
                        }
                    }

                    DirList(d);
                }
            }
            catch (System.Exception ex)
            {
                Resolver.Log.Trace(ex.Message);
            }
        }
```

doingnz, 2023-10-29:
Conversation with example screenshots showing debugging working after PDF files deleted. 
https://wildernesslabspublic.slack.com/archives/CQWD660E8/p1695850861814629

alexischr, 2023-11-28:
Huge if true!! But PDBs on-device are also useful in that they help enrich any printed-on-console stacktrace when Mono is run with "--debug".  We may still make it not the default for Debug config though.

doingnz, 2023-11-29:
I look forward to when there is the option to stop deploying the PDBs as it will significantly speed up launching the application when debugging and building all the assemblies/libraries locally in addition to saving space.  

When is Mono run with "--debug"?  When application launched from VS2022 for everyone? Or is that a special mode only for insiders?

doingnz, 2023-12-08:
Please consider the other files uploaded that possibly do not need to be pushed to Meadow.
e.g.  App.deps.json? Is that relevant to Mono in Meadow F7?

```
1>Transfer Complete, wrote 704 bytes to Meadow
1>
1>Sending file: C:\Users\centr\source\repos\Read_BATT_V\bin\Debug\netstandard2.1\App.deps.json
1>Starting File Transfer...
```

duduita, 2023-12-29:
I'm not sure if the `App.deps.json` is not used. Anyway, the CLI v1 will not include the PDB files by default, to include them it'll be necessary to use the `--includePdbs` flag during the deployment. Something similar should be done in the CLI v2.

duduita, 2024-01-18:
After changing the PDBs inclusion default to `false` on the Meadow VSCode extension, I realized that the debugger was not working properly @doingnz.

So, I believe that the debugger on the host machine may have cached or retained symbol information in memory. In that case, this cached information may still be available for setting breakpoints even though the actual PDB files have been removed during the app `Initialize` method, as you did.

doingnz, 2024-01-18:
Are you certain the debugger not working is related to not deploying PDB? Last night I was setting and clearing breakpoints in VS2022 and they were ignored.  I could see the Log.Trace output and it was clear the Meadow CCM was executing the methods with the breakpoints, yet it did not stop. The breakpoints suddenly started to work when I switched to project reference in place of nuget.

Later I removed the project references and went back to nuget and the breakpoints continued to work. I have changed projects and the breakpoints continue to work... 

Why with current VS2022 that deploys PDBs would it ignore multiple breakpoints?

doingnz, 2024-01-18:
Maybe using PDBs on the target is a mono thing I don't understand as I am not familiar with mono. My previous experience is .Net Micro Framework that can single step, watch etc C# in an embedded system and there is no file system on the target and no PDBs. 

I would expect the PDBs are always cached in the PC or available on disk locally given the debug session is launched from the PC. 

How/Why is it OK to OTA a new application only using DLLs that can be entirely different or similar and leave out of date PDBs on the target Meadow if mono is using the PDBs?

duduita, 2024-01-19:
> Are you certain the debugger not working is related to not deploying PDB?

I tested the same setup deploying the PDB files and the breakpoints worked, after not including them, the breakpoints didn't hit anymore. Another engineer tested this [PR](https://github.com/WildernessLabs/VSCode_Meadow_Extension/actions/runs/7504354043) CI build, and got the same result. So, I'm pretty confident that not including PDBs is affecting the debugger functionalities. 

> I would expect the PDBs are always cached in the PC or available on disk locally given the debug session is launched from the PC.

Anyway, I agree with you that If the debugger does take an option to load PDBs from a different place, we should use it. Remote symbols are not uncommon for debuggers to be able to do. It makes deployment twice as fast, so if there’s a way we should do it. We just need to do some research on it, after closing more P0 issues, I should take a look at it.

doingnz, 2024-01-21:
@duduita Over the weekend I built the "Bump to 1.7.0" version of the VS2022 extension. I played briefly and found that it would not stop at any breakpoints. I confirmed the PDB files were being sent to Meadow via the Log. 

Does "Bump to 1.7.0" work for other people on Visual Studio? 
Should I be able to just rebuild the project and instal the vsix file from Debug folder? 
Do I need to use the Release build or is Debug OK? 

Version 1.5.0.0 of the extension in NuGet works on my PC and stops at Meadow breakpoints.

duduita, 2024-01-23:
I haven't tested the VS2022, just the VSCode extension one.

> Over the weekend I built the "Bump to 1.7.0" version of the VS2022 extension. I played briefly and found that it would not stop at any breakpoints. I confirmed the PDB files were being sent to Meadow via the Log.
> 
> Does "Bump to 1.7.0" work for other people on Visual Studio?

Perhaps @CartBlanche should have more context about that, I'm not aware of any existent debugger problem on the VS2022. Just to confirm, have you tried to install the extension with your local build, right? In that case, have you also tried to use the breakpoints using the latest released Meadow VS2022 extension version? @doingnz

doingnz, 2024-01-23:
I can use breakpoints successfully with VS2022 and version 1.5.0 of the extension which is the latest on NuGet. 
I am curious why VS2022 thinks version 1.5.0 is the latest version when there are 1.6.0 and 1.7.0 commits in GitHub? 

When I installed local DEBUG build of the extension, specifically "Bump to 1.7.0" version in GitHub, it was not stopping execution at the breakpoints. I uninstalled and reinstalled twice to check and looked in VS2022 extension manager to confirm version listed as 1.7.0. When I uninstalled 1.7.0 and installed 1.5.0 it worked again and stopped at breakpoints.

doingnz, 2024-01-23:
Apologies ... version "Bump to 1.7.0" of the extension is working and does stop at breakpoints. I overlooked a change I introduced when reverting back to "Bump to 1.70".

When I get a chance I will play with it again to try understand why the mono debugger wants the PDB pushed to the device, but is OK for the PDB files to be deleted after execution started. 

Note the Market Place is advertising v 1.5.0 as the latest version of the extension: https://marketplace.visualstudio.com/items?itemName=WildernessLabs.vsmeadow2022

alexischr, 2025-04-16:
After execution has started, it is likely that the PDBs were already loaded in memory. Closing this issue as it seems debated and resolved.

closed 2025-04-16 by alexischr