WildernessLabs/Meadow_Issues #378: Frequent crashes after deploying app to a cloud provisioned board from Visual Studio 2022

DarkNoir, 2023-10-30:
**Describe the bug**
When MeadowCloud provisioned, there are frequent hiccups after deploying to the F7 that prevent the board from starting up.

**To Reproduce**
Steps to reproduce the behavior:
Deploy an application.
When it gets to the part where its restarted and booting, it frequently has issues here:
09:06:18:144	Mcp_2 up
09:06:24:899	Mcp_Version up
09:06:30:597	failed to parse private key -15872
09:06:36:381	mono_fdhandle_insert: duplicate Socket fd 32
09:06:37:754	mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
09:06:37:754	code '142'" (142)
09:06:54:514	mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
09:06:54:514	code '142'" (142)

I have seen other issues, one was something about and encryption key and that the board may not be provisioned. Always in the same part of the sequence.

**Expected behavior**
Board should boot up and get past all the stuff I have no control over and only crash if it gets to my bad code.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11
 - IDE and version: Microsoft Visual Studio Enterprise 2022 (64-bit) - Version 17.7.6
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v1.3.4]

**Meadow (please complete the following information as best as you can):**
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 2B-00-46-00-07-51-38-34-35-31-38-33
    Serial number: 337E316A3438
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 94:B9:7E:91:F5:FC

Firmware Versions
    OS: 1.3.4.0
    Mono: 1.3.4.0
    Coprocessor: 1.3.4.0
    Protocol: 7

**Additional context**
Started happening after provisioning for Meadow.Cloud

duduita, 2023-10-30:
Just a question, are you trying to use a client certificate for TLS authentication, i.e., have you flashed a `private_key.pem` into your device? @DarkNoir

DarkNoir, 2023-10-30:
No, no client keys nor any installed certs of any kind on the device

William Wallace
Technology Leader
William Wallace | LinkedIn<https://www.linkedin.com/in/wwallace/>
***@***.***

duduita, 2023-10-30:
I'd appreciate it if you could add the full log to the issue, just to confirm that there isn't a message like`private key file found` in the meadow console output.

DarkNoir, 2023-10-30:
At the moment its frozen at that point:

14:12:02:269	Device is configured to use WiFi for the network interface
14:12:02:657	 Mono is enabled
14:12:02:657	 Mono has been disabled - restarting Meadow
14:12:06:687	 Mono is disabled
14:12:08:764	 Mono is disabled
14:12:08:766	 dev/
14:12:08:766	 little0 [block]
14:12:08:766	 little0p0 [block]
14:12:08:766	 monostderr [char]
14:12:08:767	 monostdout [char]
14:12:08:767	 mtdblock0 [block]
14:12:08:767	 null [char]
14:12:08:767	 nxupd [char]
14:12:08:767	 pwm0 [char]
14:12:08:767	 ramlog [char]
14:12:08:767	 random [char]
14:12:08:767	 ttyACM0 [char]
14:12:08:767	 ttyS0 [char]
14:12:08:767	 ttyS1 [char]
14:12:08:767	 ttyS2 [char]
14:12:08:767	 tun [char]
14:12:08:767	 upd [char]
14:12:08:768	 urandom [char]
14:12:08:768	 usrsock [char]
14:12:08:768	 meadow0/
14:12:08:772	 ./
14:12:08:772	 ../
14:12:08:772	 App.dll [file]
14:12:08:772	 App.pdb [file]
14:12:08:772	 app.config.yaml [file]
14:12:08:772	 App.deps.json [file]
14:12:08:772	 meadow.config.yaml [file]
14:12:08:772	 Meadow.Contracts.dll [file]
14:12:08:772	 Meadow.dll [file]
14:12:08:772	 Meadow.F7.dll [file]
14:12:08:772	 Meadow.Logging.dll [file]
14:12:08:772	 Meadow.Units.dll [file]
14:12:08:772	 Microsoft.Bcl.AsyncInterfaces.dll [file]
14:12:08:772	 Microsoft.Extensions.Configuration.Abstractions.dll [file]
14:12:08:772	 Microsoft.Extensions.Configuration.dll [file]
14:12:08:772	 Microsoft.Extensions.Configuration.FileExtensions.dll [file]
14:12:08:772	 Microsoft.Extensions.Configuration.Json.dll [file]
14:12:08:773	 Microsoft.Extensions.FileProviders.Abstractions.dll [file]
14:12:08:773	 Microsoft.Extensions.FileProviders.Physical.dll [file]
14:12:08:773	 Microsoft.Extensions.FileSystemGlobbing.dll [file]
14:12:08:773	 Microsoft.Extensions.Primitives.dll [file]
14:12:08:774	 Mono.Security.dll [file]
14:12:08:774	 MQTTnet.dll [file]
14:12:08:775	 mscorlib.dll [file]
14:12:08:775	 mscorlib.pdb [file]
14:12:08:775	 NetEscapades.Configuration.Yaml.dll [file]
14:12:08:775	 netstandard.dll [file]
14:12:08:775	 System.Buffers.dll [file]
14:12:08:775	 System.Configuration.dll [file]
14:12:08:775	 System.Core.dll [file]
14:12:08:778	 System.Core.pdb [file]
14:12:08:778	 System.dll [file]
14:12:08:778	 System.IO.Compression.dll [file]
14:12:08:778	 System.IO.Compression.FileSystem.dll [file]
14:12:08:779	 System.Memory.dll [file]
14:12:08:779	 System.Net.Http.dll [file]
14:12:08:779	 System.Numerics.dll [file]
14:12:08:779	 System.Numerics.Vectors.dll [file]
14:12:08:779	 System.pdb [file]
14:12:08:779	 System.Runtime.CompilerServices.Unsafe.dll [file]
14:12:08:779	 System.Text.Encodings.Web.dll [file]
14:12:08:779	 System.Text.Json.dll [file]
14:12:08:779	 System.Web.dll [file]
[... 111 lines cut ...]
System.Text.Encodings.Web.dll                                35 kB
System.Text.Json.dll                                        397 kB
System.Web.dll                                               97 kB
System.Xml.dll                                              694 kB
YamlDotNet.dll                                               81 kB
System.Threading.Tasks.Extensions.dll                         5 kB
Bh1750.dll                                                    9 kB
Bme68x.dll                                                   24 kB
Bmi270.dll                                                   18 kB
Mcp23xxx.dll                                                 15 kB
Meadow.Foundation.dll                                        37 kB
MicroGraphics.dll                                            39 kB
ProjectLab.dll                                               17 kB
Sc16is7x2.dll                                                14 kB
TftSpi.dll                                                   13 kB
Meadow.LogProviders.dll                                      10 kB
meadow.log                                                  152 bytes
dns.conf                                                     38 bytes

Summary:
        56 files
        10.86MB of file data
        Spanning 2682 blocks
        Consuming 10.99MB on disk
Done!

duduita, 2023-11-02:
I couldn't reproduce your issue using 1.3.4.0 OS in my setup. I could provision my device and use Meadow Cloud well.

Just to give more context to you, based on your logs it seems that it found in some way a `private_key.pem` and it failed to parse it:
`failed to parse private key -15872`
It happens when someone tries to use a client certificate authentication with a bad format private key, for example.

So, I'd appreciate if you could give me more information, maybe a sample app that would allow me to reproduce your issue. Besides that, I think that might be worth trying to erase your device, you can do it using the CLI command `meadow flash erase`.

DarkNoir, 2023-11-02:
Again, there are no private keys in play that I have created. Its intermittent when the meadow is booting. I listed all files on the Meadow above.(To prove there are none)  It seems to always happen right after "09:06:24:899 Mcp_Version up". I have also had the board lock up at that same spot with no communicated errors. I was thinking it was one of your keys, perhaps to secure communications with meadow cloud? It will prevent the Meadow from starting and can only be hard reset at that point. You have all the info I have. It is intermittent so will be no fun to reproduce.  Here is my project that it was happening in. It was during the App.Initialize at:             _ProjectLab = ProjectLab.Create(); and never makes it to my next line of code.
Hardware is:
Meadow F72.c on a ProjectLab V2.e. 
[MeadowHttpClientIssue.zip](https://github.com/WildernessLabs/Meadow_Issues/files/13232965/MeadowHttpClientIssue.zip)

DarkNoir, 2023-11-02:
This happens also at that same point:
09:05:50:540	Instantiating Project Lab v2 specific hardware
09:06:18:144	Mcp_2 up
09:06:24:899	Mcp_Version up
09:06:30:597	failed to parse private key -15872
09:06:36:381	mono_fdhandle_insert: duplicate Socket fd 32
09:06:37:754	mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
09:06:37:754	code '142'" (142)
09:06:54:514	mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
09:06:54:514	code '142'" (142)

DarkNoir, 2023-11-02:
heads up, updated to 1.4.0.3

DarkNoir, 2023-11-02:
Just happened again much farther down the cycle,:

18:05:49:656	 Meadow successfully started MONO
18:05:49:657	Initializing OS...
18:05:49:657	Parsing app.config.yaml...
18:05:49:657	Log level: Information
18:05:49:657	Device is configured to use WiFi for the network interface
18:05:49:657	Update Service is enabled.
18:05:49:657	Health Metrics enabled with interval: 10 minute(s).
18:05:51:024	 Mono is enabled
18:05:51:024	 Mono has been disabled - restarting Meadow
18:05:55:055	 Mono is disabled
18:05:57:081	 Mono is disabled
18:05:57:108	 dev/
18:05:57:108	 little0 [block]
18:05:57:108	 little0p0 [block]
18:05:57:108	 monostderr [char]
18:05:57:108	 monostdout [char]
18:05:57:109	 mtdblock0 [block]
18:05:57:109	 null [char]
18:05:57:109	 nxupd [char]
18:05:57:109	 pwm0 [char]
18:05:57:109	 ramlog [char]
18:05:57:109	 random [char]
18:05:57:109	 ttyACM0 [char]
18:05:57:109	 ttyS0 [char]
18:05:57:109	 ttyS1 [char]
18:05:57:109	 ttyS2 [char]
18:05:57:109	 tun [char]
18:05:57:110	 upd [char]
18:05:57:110	 urandom [char]
18:05:57:110	 usrsock [char]
18:05:57:110	 meadow0/
18:05:57:110	 ./
18:05:57:111	 ../
18:05:57:111	 Cache/
18:05:57:114	 ./
18:05:57:114	 ../
18:05:57:114	 Data/
18:05:57:117	 ./
18:05:57:117	 ../
18:05:57:117	 Documents/
18:05:57:120	 ./
18:05:57:120	 ../
18:05:57:120	 cloud.log [file]
18:05:57:120	 events.log [file]
18:05:57:120	 Temp/
18:05:57:123	 ./
18:05:57:123	 ../
18:05:57:123	 update-store/
18:05:57:126	 ./
18:05:57:126	 ../
18:05:57:126	 App.dll [file]
18:05:57:127	 App.pdb [file]
18:05:57:127	 app.config.yaml [file]
18:05:57:127	 App.deps.json [file]
18:05:57:127	 meadow.config.yaml [file]
18:05:57:127	 Mcp23xxx.dll [file]
18:05:57:127	 Meadow.Contracts.dll [file]
18:05:57:127	 Meadow.dll [file]
18:05:57:128	 Meadow.F7.dll [file]
18:05:57:128	 Meadow.Foundation.dll [file]
18:05:57:128	 Meadow.Logging.dll [file]
18:05:57:135	 Meadow.LogProviders.dll [file]
18:05:57:135	 Meadow.Units.dll [file]
18:05:57:136	 MicroGraphics.dll [file]
18:05:57:136	 Microsoft.Bcl.AsyncInterfaces.dll [file]
18:05:57:136	 Microsoft.Extensions.Configuration.Abstractions.dll [file]
18:05:57:136	 Microsoft.Extensions.Configuration.dll [file]
18:05:57:136	 Microsoft.Extensions.Configuration.FileExtensions.dll [file]
18:05:57:136	 Microsoft.Extensions.Configuration.Json.dll [file]
18:05:57:136	 Microsoft.Extensions.FileProviders.Abstractions.dll [file]
18:05:57:137	 Microsoft.Extensions.FileProviders.Physical.dll [file]
18:05:57:137	 Microsoft.Extensions.FileSystemGlobbing.dll [file]
18:05:57:137	 Microsoft.Extensions.Primitives.dll [file]
18:05:57:137	 Mono.Security.dll [file]
18:05:57:137	 MQTTnet.dll [file]
18:05:57:137	 mscorlib.dll [file]
18:05:57:137	 mscorlib.pdb [file]
18:05:57:137	 NetEscapades.Configuration.Yaml.dll [file]
18:05:57:137	 netstandard.dll [file]
18:05:57:137	 ProjectLab.dll [file]
18:05:57:138	 Sc16is7x2.dll [file]
18:05:57:138	 System.Buffers.dll [file]
18:05:57:138	 System.Configuration.dll [file]
18:05:57:138	 System.Core.dll [file]
18:05:57:138	 System.Core.pdb [file]
18:05:57:138	 System.dll [file]
18:05:57:138	 System.IO.Compression.dll [file]
18:05:57:138	 System.IO.Compression.FileSystem.dll [file]
18:05:57:138	 System.Memory.dll [file]
18:05:57:138	 System.Net.Http.dll [file]
18:05:57:138	 System.Numerics.dll [file]
18:05:57:139	 System.Numerics.Vectors.dll [file]
18:05:57:139	 System.pdb [file]
18:05:57:139	 System.Runtime.CompilerServices.Unsafe.dll [file]
18:05:57:139	 System.Text.Encodings.Web.dll [file]
18:05:57:139	 System.Text.Json.dll [file]
18:05:57:139	 System.Web.dll [file]
18:05:57:140	 System.Xml.dll [file]
18:05:57:140	 TftSpi.dll [file]
18:05:57:140	 YamlDotNet.dll [file]
18:05:57:140	 System.Threading.Tasks.Extensions.dll [file]
18:05:57:140	 meadow.log [file]
18:05:57:140	 dns.conf [file]
18:05:57:140	 var/
18:05:57:141	 mqueue/
18:05:57:141	 Esp32Events [char]
18:05:57:141	 Esp32Requests [char]
18:05:57:141	 IncomingEvents [char]
18:06:11:504	 Meadow successfully deleted '/meadow0/meadow.log'
18:06:11:582	 Meadow successfully deleted '/meadow0/dns.conf'
18:06:11:819	 Download of 'wifi.config.yaml' success (checksums calculated:0xE534C112, expected:0xE534C112)
18:06:11:824	 Mono is disabled
18:06:11:826	 Mono has been enabled - restarting Meadow
18:06:13:844	 Meadow successfully started MONO
18:06:15:871	 Mono is enabled
18:06:20:135	Initializing OS...
18:06:22:212	Parsing app.config.yaml...
18:06:24:076	Log level: Information
18:06:27:022	Device is configured to use WiFi for the network interface
18:06:32:712	Update Service is enabled.
18:06:33:147	Health Metrics enabled with interval: 10 minute(s).
18:06:33:519	Initialize...
18:06:35:949	Instantiating Project Lab v2 specific hardware
18:06:48:048	Mcp_2 up
18:06:48:247	Mcp_Version up
18:06:51:064	Project Lab Board Initialized
18:06:51:236	Environment Sensor Initialized
18:07:05:916	Display Initialized
18:07:10:756	Run...
18:07:11:175	Info loop started
18:07:12:262	Starting Send .......
18:07:16:491	Data loop took more than a second 00:00:04.6590000
18:07:19:063	Posting data.......
18:07:19:221	failed to parse private key -15872

DarkNoir, 2023-11-02:
The attached project will always fail at some point. It was designed to try to figure out the issues with HttpCLient, TLS, and the POST method. It exposes a memory leak and will eventually crash or lock up.

duduita, 2023-11-02:
> Again, there are no private keys in play that I have created. Its intermittent when the meadow is booting. I listed all files on the Meadow above.(To prove there are none)

What I meant is that it should not be a problem with a meadow cloud, since this error message is related to our TLS provider code, especially with the client certificate auth implementation, but I got your point that you don't have a `private_key.pem` stored in your device. Anyway, I have an idea of what it might be, but I need to reproduce the issue first.

Looking at your [MeadowHttpClientIssue.zip](https://github.com/WildernessLabs/Meadow_Issues/files/13232965/MeadowHttpClientIssue.zip) it seems that this zip doesn't contain the entire project code.

DarkNoir, 2023-11-02:
let me check, I did mess with it a bit

DarkNoir, 2023-11-02:
[MeadowHttpClientIssue.zip](https://github.com/WildernessLabs/Meadow_Issues/files/13233380/MeadowHttpClientIssue.zip)
Yeah, the project folder got deselected when i zipped it

DarkNoir, 2023-11-02:
The error has happened during initialization before I make any network calls.

duduita, 2023-11-02:
Thank you @DarkNoir, I'll take a look at it

DarkNoir, 2023-11-02:
I am also on you Slack instance "William Wallace"  william.wallace@wallsoft-inc.com

DarkNoir, 2023-11-02:
internal meadow cloud apis do not use TLS?

DarkNoir, 2023-11-02:
The endpoint the application is using is safe for you to use for testing also. Feel free to use it, it is an azure function that just echos the json payload

duduita, 2023-11-02:
> internal meadow cloud apis do not use TLS?

It uses, but should return an error slightly different if the meadow cloud private key fails to be parsed.

duduita, 2023-11-06:
Thanks for your help @DarkNoir, I could reproduce your issue, and make a fix for it. After running many times, it seems working well now, and should be fixed in the upcoming release.

DarkNoir, 2023-11-06:
Thats great news. I can't wait till this little puppy is stable enough to use in a production environment.  Keep an eye out for anything that's causing or can cause memory leaks. I have discovered that if I don't force a GC frequently. memory usage continues to climb until the board stops working. 

What is your release cadence?

closed 2024-01-17 by duduita