WildernessLabs/Meadow_Issues #379: HttpClient memory leak when using HTTPS

DarkNoir, 2023-10-30:
**Describe the bug**
There is a memory leak when using the HttpClient, HTTPS and POST requests. It does not occur when using HTTP.

**To Reproduce**
Steps to reproduce the behavior:
1. HttpClient can't be used for multiple requests (an issue has been reported for that) so you will need to create a new HttpClient for each request. I attached the solution I was using to surface the issue. Basically, create an app that makes repeated HTTPS POST calls using HttpClient. 
2. Observer memory usage increase steadily.

**Expected behavior**
I would expect to be able to use a single instance of HttpClient for repeated calls.
I would expect memory usage would not increase to the point where the app runs out of memory.

**Screenshots**
[image]

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11
 - IDE and version: Microsoft Visual Studio Enterprise 2022 (64-bit) -Version 17.7.6
 - Meadow extension for IDE version: 1.3.4

**Meadow (please complete the following information as best as you can):**
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 2B-00-46-00-07-51-38-34-35-31-38-33
    Serial number: 337E316A3438
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 94:B9:7E:91:F5:FC

Firmware Versions
    OS: 1.3.4.0
    Mono: 1.3.4.0
    Coprocessor: 1.3.4.0
    Protocol: 7

**Additional context**
Project Lab v2 using the display.

DarkNoir, 2023-10-30:
[MeadowHttpClientIssue.zip](https://github.com/WildernessLabs/Meadow_Issues/files/13208560/MeadowHttpClientIssue.zip)

DarkNoir, 2023-12-15:
This looks fixed in 1.6 :)

duduita, 2023-12-19:
I'll try to reproduce it here again using OS v1.6 to confirm that. Perhaps the WiFi memory leak fix might have fixed it as well @alexischr

duduita, 2024-01-25:
> This looks fixed in 1.6 :)

After an overnight test, I couldn't reproduce this issue as well using v1.8.0.0. So, I believe that we can close this issue, since it also looks fixed to me.

[image]

closed 2024-01-25 by duduita