WildernessLabs/Meadow_Issues #380: HttpClient errors: The descriptor is not a socket

DarkNoir, 2023-10-31:
**Describe the bug**
I have an F7 1.3.4 that is Meadow.Cloud provisioned with updates and health turned on sitting on a Project Lab v2 and I am sending data via **Https POST every 10 seconds**. I am seeing a number of these errors:

19:15:40:752	Posting data.......
19:15:42:049	System.Net.Http.HttpRequestException: The descriptor is not a socket ---> System.Net.Sockets.SocketException: The descriptor is not a socket
19:15:42:049	at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) [0x00110] in <218039eb271b41ce9d93d97b1d4fb99a>:0
19:15:42:049	--- End of inner exception stack trace ---
19:15:42:049	a
19:15:42:099	System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) [0x001ac] in <218039eb271b41ce9d93d97b1d4fb99a>:0
19:15:42:100	at System.Net.Http.HttpConnectionPool.CreateConnectionAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00134] in <218039eb271b41ce9d9
19:15:42:100	d97b1d4fb99a>:0
19:15:42:100	at System.Net.Http.HttpConnectionPool.WaitForCreatedConnectionAsync (System.Threading.Tasks.ValueTask`1[TResult] creationTask) [0x000a2] in <218039eb271b41ce9d93d97b1d4fb99a>:0
19:15:42:100	at System.Net.Http.HttpConnectionPool.SendWithRetryAsync (System.Net.Http.HttpRequestMessage request, System.Boolean doRequestAuth, System.Threading.CancellationToken cancellationToken
19:15:42:151	[0x00089] in <218039eb271b41ce9d93d97b1d4fb99a>:0
19:15:42:151	at System.Net.Http.RedirectHandler.SendAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00094] in <218039eb271b41ce9d93d97b1d4fb99a>:0
19:15:42:151	at System.Net.Http.HttpClient.FinishSendAsyncBuffered (System.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequestM
19:15:42:151	ssage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) [0x0016c] in <218039eb271b41ce9d93d97b1d4fb99a>:0
19:15:42:151	at MeadowHttpClientIssue.MeadowApp.SendData (MeadowHttpClientIssue.SensorDataMessage[] data) [0x0013d] in <758e0d0
19:15:42:153	993f14f1bae4503e06c860074>:0
19:15:42:155	SendData Completed...

ALSO:

System.Net.Http.HttpRequestException: Not enough buffer space is available ---> System.Net.Sockets.SocketException: Not enough buffer space is available
13:39:37:367	at System.Net.Sockets.Socket..ctor (System.Net.Sockets.AddressFamily addressFamily, System.Net.Sockets.SocketType socketType, System.Net.Sockets.ProtocolType protocolType) [0x00069] in <142ee1e4e94e4c4791b02f916dc163d6>:0
13:39:37:367	at
13:39:37:367	ystem.Net.Sockets.DualSocketMultipleConnectAsync..ctor (System.Net.Sockets.SocketType socketType, System.Net.Sockets.ProtocolType protocolType) [0x0000d] in <142ee1e4e94e4c4791b02f916dc163d6>:0
13:39:37:367	at System.Net.Sockets.Socket.ConnectAsync (System.Net.Sockets.SocketType socketType, System.Net.Sockets.ProtocolType protocolType, System.Net.Sockets.SocketAsyncEventArgs e) [0x00056] in
13:39:37:367	<142ee1e4e94e4c4791b02f916dc163d6>:0
13:39:37:367	at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.CancellationToken cancellationToken) [0x00058] in <218039eb271b41ce9d93d97b1d4fb99a>:0
13:39:37:367	--- End of inner exception stack trace ---
13:39:37:367	at System.Net.Http.ConnectHelper.ConnectAsync (System.String host, System.Int32 port, System.Threading.C
13:39:37:367	ncellationToken cancellationToken) [0x001ac] in <218039eb271b41ce9d93d97b1d4fb99a>:0
13:39:37:367	at System.Net.Http.HttpConnectionPool.CreateConnectionAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00134] in <218039eb271b41ce9d93d97b1d4fb99a>:0
13:39:37:367	at System.Net.Http.HttpConnectionPool.WaitForCreatedConnectionAsync (System.Threadi
13:39:37:367	g.Tasks.ValueTask`1[TResult] creationTask) [0x000a2] in <218039eb271b41ce9d93d97b1d4fb99a>:0
13:39:37:367	at System.Net.Http.HttpConnectionPool.SendWithRetryAsync (System.Net.Http.HttpRequestMessage request, System.Boolean doRequestAuth, System.Threading.Cancellati
13:39:37:377	onToken cancellationToken) [0x00089] in <218039eb271b41ce9d93d97b1d4fb99a>:0
13:39:37:377	at System.Net.Http.RedirectHandler.SendAsync (System.Net.Http.HttpRequestMessage request, System.Threading.CancellationToken cancellationToken) [0x00094] in <218039eb271b41ce9d9
13:39:37:379	3d97b1d4fb99a>:0
13:39:37:379	at System.Net.Http.HttpClient.FinishSendAsyncBuffered (System.Threading.Tasks.Task`1[TResult] sendTask, System.Net.Http.HttpRequestMessage request, System.Threading.CancellationTokenSource cts, System.Boolean disposeCts) [0x0016c] in <218039eb271b41ce9d93d97b1d4fb99a>:0
13:39:37:379	at MeadowHttpClientIssue.MeadowApp.<SendData>b__16_0 () [0x00237] in <5cb8b87e94764a0ea88
13:39:37:379	7b0cece0a6d9>:0

**To Reproduce**
Steps to reproduce the behavior:
Attaching the project I am running. Deploy the app and let it run a few minutes.

**Expected behavior**
Expected completed http post calls 

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11
 - IDE and version: Microsoft Visual Studio Enterprise 2022 (64-bit) - Version 17.7.6
 - Meadow extension for IDE version: 1.3.4

**Meadow (please complete the following information as best as you can):**
Meadow by Wilderness Labs
Board Information
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: MeadowF7

Hardware Information
    Processor type: STM32F777IIK6
    ID: 2B-00-46-00-07-51-38-34-35-31-38-33
    Serial number: 337E316A3438
    Coprocessor type: ESP32
    MAC Address -
        WiFi: 94:B9:7E:91:F5:FC

Firmware Versions
    OS: 1.3.4.0
    Mono: 1.3.4.0
    Coprocessor: 1.3.4.0
    Protocol: 7

**Additional context**
I was using this project to research the memory leak issues I was having with HttpClient calls and started having these issues. Originally, I was doing one call every 60 seconds. First call always takes around 30 seconds. Subsequent calls take about 3 seconds.  Quite a bit slower than I would expect since the same calls from my PC take around 20-100ms. Based on this so far, a sensor resolution of 1 second would be unobtainable. 
[MeadowHttpClientIssue-httperrors.zip](https://github.com/WildernessLabs/Meadow_Issues/files/13211446/MeadowHttpClientIssue-httperrors.zip)

DarkNoir, 2023-11-06:
I concur.

William Wallace
Technology Leader
William Wallace | LinkedIn<https://www.linkedin.com/in/wwallace/>
***@***.***

I think that this issue is related to the issue #355<https://github.com/WildernessLabs/Meadow_Issues/issues/355>.

duduita, 2023-12-06:
FYI, using Ethernet, after some hours of testing, I couldn't reproduce these errors. They seem to only happen on WiFi, so I'll focus on investigating the coprocessor, which is responsible for the WiFi, to understand why it's happening.

alexischr, 2024-01-16:
Fixed in OS 1.8

closed 2024-01-16 by alexischr