WildernessLabs/Meadow_Issues #381: HttpClient.GetAsync Run Concurrently with MapleServer Crashes

E-R-T-W, 2023-10-31:
### **Describe the bug**
Run a Task to get JSON data in a loop concurrently with MapleServer running and intermittently, but without fail, the board will crash with `mono_coop_mutex_lock Cannot transition thread 0x1e from STATE_BLOCKING with DO_BLOCKING` 

### **To Reproduce**
Steps to reproduce the behavior:
1. Deploy the attached code to the Meadow F7v2 Feather. The code will start a task making HttpClient.GetAsync requests for JSON data, while concurrently starting a MapleServer
2. From a web browser, navigate to http://192.168.1.169:5417/Led/state/ON to send a command to turn on an LED via the MapleServer. (One can also turn off the LED via http://192.168.1.169:5417/Led/state/OFF, and also request Analog Input Voltages via http://192.168.1.169:5417/Led/state/A00).
3. Intermittently, but eventually, without fail, if you keep issuing commands to the MapleServer as described in Step 2, the board will crash with `mono_coop_mutex_lock Cannot transition thread 0x1e from STATE_BLOCKING with DO_BLOCKING`, I use a Watchdog Timer to Reset the board upon a crash.
4. To easily, quickly, reliably re-create the bug, simply mash on the refresh button in a web browser.

### **Expected behavior**
You should be able to make JSON data requests via HttpClient.GetAsync while concurrently running and issuing commands to the MapleServer.

### **Screenshots**
Source Code:
[SourceCode.zip](https://github.com/WildernessLabs/Meadow_Issues/files/13211848/SourceCode.zip)

### **Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11 Pro v22H2
 - IDE and version: Visual Studio Professional 2022 v17.7.5
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v1.3.4

### **Meadow (please complete the following information as best as you can):**
Most of these values can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7v2
 - Meadow OS version: 1.3.4.0
 - If they are different, please provide these versions as well.
   - Meadow Mono version: 1.3.4.0
   - Meadow co-processor version: 1.3.4.0

### **Additional context**
Much discussion with regards to this issue has been posted on Slack. For example see:
https://wildernesslabspublic.slack.com/archives/CQWD660E8/p1695877053007039
brian.kim@wildernesslabs.co](mailto:brian.kim@wildernesslabs.co has been able to reproduce similar results - see issue #355

E-R-T-W, 2023-11-03:
This is exact same issue as https://github.com/WildernessLabs/Meadow_Issues/issues/383 for Sockets - same problem.

E-R-T-W, 2023-11-03:
Suspect perhaps issue #286 might have something to do with it too. Or maybe at same time can resolve #286.

halyssonJr, 2024-01-17:
Reproducing this issue using the OS `v1.7.0.0`, clearly, we still had crashes on F7. But upload the new version OS `v1.8.0.0`, it seems to work perfectly. 
- Final Result:
```
Meadow StdOut: post HTTP request #490 (number of requests)
Meadow StdOut: Maple request #485 (number of requests)
``

closed 2024-01-17 by alexischr