WildernessLabs/Meadow_Issues #383: HttpClient.GetAsync() concurrent with Sockets (TcpListner and TcpClient) crashes

E-R-T-W, 2023-11-03:
### **Describe the bug**
When running a Tasks with HttpClient.GetAsync() to pull JSON data, while concurrently using Sockets to a communicate with the Meadow F7v2 Feather, the board will crash with `mono_coop_mutex_lock Cannot transition thread 0x14 from STATE_BLOCKING with DO_BLOCKING`

### **To Reproduce**
Steps to reproduce the behavior:
1. Deploy the Listner code attached below to the Meadow F7v2 Feather
2. Run the Client Console app attached below
3. After only a short time you will get the error `mono_coop_mutex_lock Cannot transition thread 0x14 from STATE_BLOCKING with DO_BLOCKING` and HttpClient.GetAsync() will no longer be able to run to pull JSON data

### **Expected behavior**
One should be able to run HttpClient.GetAsync() concurrently with Sockets (TcpListner and TcpClient).

### **Screenshots**
Meadow F7v2 Feather Source Code:
[Listener.zip](https://github.com/WildernessLabs/Meadow_Issues/files/13254112/Listener.zip)

Client Console App Source Code:
[Client.zip](https://github.com/WildernessLabs/Meadow_Issues/files/13254117/Client.zip)

### **Developer tools (please complete the following information as best as you can):**
 - OS and version: Windows 11 Pro v22H2
 - IDE and version: Visual Studio Professional 2022 v17.7.5
 - Meadow extension for IDE version: VS 2022 Tools for Meadow v1.4.0

### **Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: F7v2
 - Meadow OS version: 1.4.0.3
 - If they are different, please provide these versions as well.
   - Meadow Mono version: 1.4.0.3
   - Meadow co-processor version: 1.4.0.3

### **Additional context**
This is exact same issue as #381 for MapleServer - same problem.

bryancostanich, 2023-11-03:
Same bug as #381

E-R-T-W, 2023-11-03:
Suspect perhaps issue #286 might have something to do with it too. Or maybe at same time can resolve #286.

duduita, 2024-01-15:
I could reproduce this issue using the OS v1.4.0.3 indeed, but when I tested it in newer versions, like v1.7.0.0, it seemed working well. Have you tried it in a new version, like v.1.7.0.0 or higher @E-R-T-W ? I haven't made long tests yet, but for now, I can make thousands of requests and it still keeps going. Perhaps, it has been already fixed.

E-R-T-W, 2024-01-15:
Yes I’m using 1.7.0.0 – I’ll upload a new source example of the problem for you.

Thanks for checking into it!

E-R-T-W, 2024-01-15:
I replied to Eduardo Menezes's email stating still a problem in version 1.7.0.0, but perhaps it's a different case. I told him I'd upload new source of the problem, but if it's closed, I'll create a new issue with new source example of the essentially the same problem.

Thanks for looking into it.

Closed #383<https://github.com/WildernessLabs/Meadow_Issues/issues/383> as completed.

-

alexischr, 2024-01-16:
Awaiting for @E-R-T-W 's reproduction, but the issue does not seem to be fixed. @E-R-T-W have you tried re-deploying your app if you have not done so since upgrading the OS? We have changes in the .NET DLL assemblies that may be crucial to this being fixed.

duduita, 2024-01-17:
I tried again to reproduce it, so I'd like to share some logs that I got using the OS `1.8.0.0` and `1.4.3.0`. 

As can be seen in the [1.4.0.3_listener.txt](https://github.com/WildernessLabs/Meadow_Issues/files/13956981/1.4.0.3_listener.txt) and [1.4.0.3_client.txt](https://github.com/WildernessLabs/Meadow_Issues/files/13956983/1.4.0.3_client.txt) files, I'm getting the transition issues after a short time. However, using the `1.8.0.0` I could let it run for much more time without seeing this error, until I decided to finish the application, as can be seen in the [1.8.0.0_listener.txt](https://github.com/WildernessLabs/Meadow_Issues/files/13956978/1.8.0.0_listener.txt) and [1.8.0.0_client.txt](https://github.com/WildernessLabs/Meadow_Issues/files/13956987/1.8.0.0_client.txt) files.

Perhaps, @E-R-T-W, you just need to re-deploy your app as Alexis suggested, since it will update the .NET DLL assemblies.

E-R-T-W, 2024-01-17:
Alexis, Eduardo, Brian, I tried to reproduce the error again with the latest version of my code – I was going to strip out just the relevant stuff and upload for you to test, but prior to doing that I upgraded to 1.8.0.0.

Since upgrading and flashing the OS, I can no longer reproduce the error.
I have 18 of these around the plant floor, 12 running one app, and 6 another. It could be that perhaps I hadn’t flashed one of them??? I don’t know what to say, except for thank you and my apologies for sending you on what seems to have been a wild goose chase ☹

I’ll be sure to double check they are all flashed to 1.8.0.0!

Thanks so much for fixing this bug!!! It’s fantastic to have this working!!!

Your hard work is very much appreciated!!!👊🎉🍻

closed 2024-01-17 by duduita