WildernessLabs/Meadow_Issues #391: mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error code '142'"

adrianstevens, 2023-11-14:
Running the GNSS Tracker demo app produces the following output:

```
Running App
Run...
BMI270:        X:0.0g, Y:0.0g, Z:1.0g
mono_coop_cond_wait Cannot transition thread 0x37 from STATE_BLO
CKING with DO_BLOCKING
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
BME688:        27.0C, 27%, 1016mbar
mono_coop_cond_wait Cannot transition thread 0x98 from STATE_BLO
CKING with DO_BLOCKING
mono_os_cond_wait: pthread_cond_wait failed with "Error code '14
2'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
SCD40:         27.0C, 29%, 547ppm
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
GNSS Position: lat: [49.00° 13.96' 0.00"N], long: [123.00° 10.81' 0.00"W]
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error
code '142'" (142)
Solar Voltage: 0 volts
```

adrianstevens, 2023-11-14:
Apparently happens on Clima as well. 

This suggests to me it could be a serial port related issue.

doingnz, 2023-11-23:
@adrianstevens made any progress tracking down this issue?

bryancostanich, 2024-03-14:
Wondering if this is `System.Text.Json` and is the same error as #494

adrianstevens, 2024-03-15:
this was from the round-robin testing and is fixed - closing issue

closed 2024-03-15 by adrianstevens