WildernessLabs/Meadow_Issues #400: Recent addition of subdirectories overlooked two features

Peter-Moody, 2023-12-04:
**Describe the bug**
There are 2 features that are missing in the recent addition of subdirectories
1. If CLI user sends a single '/' character it is rejected. I should not be rejected but be understood to be the root of the Meadow fIle system.
2. There is no way to execute the meadow file list command and obtain directory names.

ctacke, 2023-12-04:
For directories, I'd say just send them back with the files, but give them a CRC of something like -1 as a flag that they are directories

Peter-Moody, 2023-12-04:
The bug was reported when ESP32 data was being downloaded. But, it was being tested for both ESP32 and STM32 downloads. On investigation it was found that the bug was redundant code as this same test is reapplied properly in a different place.

closed 2024-03-05 by alexischr