WildernessLabs/Meadow_Issues #403: Add Method to Wait for Cloud Logs to sync

thegreatco, 2023-12-07:
**Is your feature request related to a problem? Please describe.**
When running in a low power environment, it is necessary to put the MCU to sleep quite often. There is currently no way to wait for logs to flush to the cloud before sleeping. In the most pathological case, logs never make it to the cloud as the MCU is never awake long enough to push.

**Describe the solution you'd like**
Some sort of `WaitForSync` or `Flush` or some other method that can be called to wait for the current batch of logs to sync. It may also be necessary to add some sort of barrier so that logs written after `WaitForSync` is called are buffered until the next call to `WaitForSync`.

**Describe alternatives you've considered**
Writing my own logging sink to control this behavior.

**Additional context**
Add any other context or screenshots about the feature request here.

alexischr, 2023-12-12:
Sounds like Meadow.Cloud sync needs to be awaited on in a BeforeSleep()

alexischr, 2024-10-01:
Maybe we need an `await Flush()` for Meadow.Cloud events/ReliabilityService