WildernessLabs/Meadow_Issues #409: Network gets unstable if a transaction times out

ctacke, 2023-12-14:
Setting a shorter timeout on a network transaction causes things to go sideways, with the OS appearing to lock up.

A concrete example is this:

```
using (var client = new HttpClient())
{
    client.Timeout = TimeSpan.FromSeconds(10);

    var json = JsonSerializer.Serialize<dynamic>(new { id = Resolver.Device.Information.UniqueID.ToUpper() });
    var content = new StringContent(json, Encoding.UTF8, "application/json");

    var endpoint = $"{Settings.Hostname}/api/devices/login";
    var response = await client.PostAsync(endpoint, content);
}
```

If this is the first communication with the end point, it will time out on the F7, as the first transaction is slow.  However, if you then re-try this same method, the application calls `PostAsync` and never returns.  

I suspect the timeout/cancellation is putting the network in a bad state - probably related to the native/managed descriptors not matching

ctacke, 2023-12-20:
Strange.  I pared it down to a simpler app and it doesn't crash in the same place (though it still dies quickly).  The attached sample app demonstrates the failure on 1.6.0.1
[network-timeout.zip](https://github.com/WildernessLabs/Meadow_Issues/files/13729942/network-timeout.zip)

bryancostanich, 2024-01-09:
We think we got this fixed with @halyssonJr's TLS fixes, going to spin new OS bins for you to validate, @ctacke.

ctacke, 2024-02-06:
Validated fixed on 1.8.0.0

closed 2024-02-06 by ctacke