WildernessLabs/Meadow_Issues #417: Linker Bug During Deployment

jeffjulian, 2024-01-02:
I noticed this error in the command log for deploying my Meadow app, and I thought I would share it.  Not sure what caused it, but you probably want to handle it somehow other than throwing an exception and bailing out.

```cmd 
Found Microsoft.Extensions.DependencyInjection.Abstractions.dll (CRC: 3811836787)

Found meadow.log (CRC: 2911356623)

Found dns.conf (CRC: 551742154)

no-link args: '-p copy "Sentinel.Isaiah.Device" -p copy "Sentinel.ApiClient" -p copy "Sentinel.Api.Models" -p copy "Sentinel.Core" -p copy "System.Text.Json" -p copy "mscorlib" -p copy "System.Core"'
Trimming assemblies to reduce size (may take several seconds)...
StandardError Contains: Process terminated. Assertion failed.
Invalid IL or a bug in the scanner
   at Mono.Linker.Dataflow.ReflectionMethodBodyScanner.WarnAboutInvalidILInMethod(MethodBody method, Int32 ilOffset) in /Users/lexas/linker/src/linker/Linker.Dataflow/ReflectionMethodBodyScanner.cs:line 171
   at Mono.Linker.Dataflow.MethodBodyScanner.Scan(MethodBody methodBody) in /Users/lexas/linker/src/linker/Linker.Dataflow/MethodBodyScanner.cs:line 559
   at Mono.Linker.Dataflow.ReflectionMethodBodyScanner.ScanAndProcessReturnValue(MethodBody methodBody) in /Users/lexas/linker/src/linker/Linker.Dataflow/ReflectionMethodBodyScanner.cs:line 69
   at Mono.Linker.Steps.MarkStep.MarkReflectionLikeDependencies(MethodBody body, Boolean requiresReflectionMethodBodyScanner) in /Users/lexas/linker/src/linker/Linker.Steps/MarkStep.cs:line 2961
   at Mono.Linker.Steps.MarkStep.MarkMethodBody(MethodBody body) in /Users/lexas/linker/src/linker/Linker.Steps/MarkStep.cs:line 2805
   at Mono.Linker.Steps.MarkStep.ProcessMethod(MethodDefinition method, DependencyInfo& reason) in /Users/lexas/linker/src/linker/Linker.Steps/MarkStep.cs:line 2428
   at Mono.Linker.Steps.MarkStep.ProcessQueue() in /Users/lexas/linker/src/linker/Linker.Steps/MarkStep.cs:line 385
   at Mono.Linker.Steps.MarkStep.ProcessPrimaryQueue() in /Users/lexas/linker/src/linker/Linker.Steps/MarkStep.cs:line 370
   at Mono.Linker.Steps.MarkStep.Process() in /Users/lexas/linker/src/linker/Linker.Steps/MarkStep.cs:line 337
   at Mono.Linker.Steps.MarkStep.Process(LinkContext context) in /Users/lexas/linker/src/linker/Linker.Steps/MarkStep.cs:line 181
   at Mono.Linker.Pipeline.ProcessStep(LinkContext context, IStep step) in /Users/lexas/linker/src/linker/Linker/Pipeline.cs:line 135
   at Mono.Linker.Pipeline.Process(LinkContext context) in /Users/lexas/linker/src/linker/Linker/Pipeline.cs:line 128
   at Mono.Linker.Driver.Run(ILogger customLogger) in /Users/lexas/linker/src/linker/Linker/Driver.cs:line 755
   at Mono.Linker.Driver.Main(String[] args) in /Users/lexas/linker/src/linker/Linker/Driver.cs:line 64

Trimming failed - ILLinker execution error!
Exit Code: -2146232797
Meadow StdInfo: Meadow successfully deleted '/meadow0/meadow.log'
Removing file: FileData { FullPath = /meadow0/meadow.log, FileName = meadow.log, Crc = 2911356623, FileSize = 152 }

Meadow StdInfo: Meadow successfully deleted '/meadow0/dns.conf'
Removing file: FileData { FullPath = /meadow0/dns.conf, FileName = dns.conf, Crc = 551742154, FileSize = 38 }
```

**To Reproduce**
Steps to reproduce the behavior:
1. Set Meadow CLI Port
2. Build .NET Project
3. Deploy with --no-link options

alexischr, 2024-02-06:
We'll need a reproduction for this issue, we have not ourselves seen it in action. 

@CartBlanche @adrianstevens question: In CLI, if anyone has trouble with the linker, they can just disable it. Do VS extension users have a similar way of working around this?

adrianstevens, 2024-03-26:
It is possible but most apps won't fit on Meadow without linking unfortunately

adrianstevens, 2026-11-07:
Closing: this was reported against the 1.x deployment linker, which has since been replaced. Please open a new issue if it reproduces on a current release.

closed 2026-11-07 by adrianstevens