WildernessLabs/Meadow_Issues #427: HTTP serial requests are resulting mono_poll error

duduita, 2024-01-15:
**Describe the bug**
When trying to make serial HTTP requests, I'm getting the `mono_poll` error code 23.

```
Meadow StdInfo: Meadow successfully started MONO
Listening for Meadow Console output. Press Ctrl+C to exit
Meadow StdOut: Initializing OS...
Meadow StdOut: Parsing app.config.yaml...
Meadow StdOut: Log level: Debug
Meadow StdOut: Device is configured to use WiFi for the network interface
Meadow StdOut: Device configured to auto-connect to SSID xxxx
Meadow StdOut: Update Service is disabled.
Meadow StdOut: Health Metrics disabled.
Meadow StdOut: 21:58:59: Connecting to router via WiFi.
Meadow StdOut: 21:59:28: WiFi connected.
Meadow StdOut: 21:59:28: IP Address: 192.168.2.116
Meadow StdOut: 21:59:28: Network connection established.
Meadow StdOut: HttpGetSoakTest
Meadow StdOut: 21:59:29 1
Meadow StdOut: 21:59:36 2
Meadow StdOut: 21:59:36 3
Meadow StdOut: 21:59:36 4
Meadow StdOut: 21:59:37 5
Meadow StdOut: 21:59:37 6
Meadow StdOut: 21:59:38 7
Meadow StdOut: 21:59:42 8
Meadow StdOut: 21:59:44 9
Meadow StdOut: poll_event_wait: mono_poll () failed, error (23) Error code '23'
```
**To Reproduce**
Steps to reproduce the behavior:
1. Checkout [Meadow TestSuite](https://github.com/WildernessLabs/Meadow.TestSuite)
2. Application - `src/Network/FeatherF7SoakTestRunner`  Make the usual changes to `wifi.config.yaml`
3. Edit `app.config.yaml` to look like this:
```
Lifecycle:
    RestartOnAppFailure: false
    AppFailureRestartDelaySeconds: 15

Logging:
    LogLevel: 
        Default: "Debug"

SoakTest:
    #
    #   Name of the test to execute.
    #
    TestName: "HttpGetSoakTest"
    #
    #   What should we request?
    #
    # RequestUri: "http://192.168.1.239/"
    RequestUri: "http://postman-echo.com/get?foo1=bar1&foo2=bar2"
    #
    #   Number of test cycles.
    #
    NumberOfCycles: 10000
    #
    #   Number of milliseconds between cycles.
    #
    DelayBetweenCyclesMs: 0

```
**Expected behavior**
It's expected to be able to do thousands of serial requests, as we can do using HTTPS.

**Developer tools (please complete the following information as best as you can):**
 - OS and version: macOS Ventura v13.2.1

**Meadow (please complete the following information as best as you can):**
Most of these values can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
```
Meadow StdInfo: Meadow successfully started MONO
Meadow by Wilderness Labs
Board Information 
    Model: F7Micro
    Hardware version: F7FeatherV2
    Device name: F7SoakTestRunner

Hardware Information 
    Processor type: STM32F777IIK6
    ID: 17-00-34-00-07-51-38-34-35-31-38-33
    Serial number: 336C31563438
    Coprocessor type: ESP32
    MAC Address - 
        WiFi: 94:B9:7E:91:0C:DC

Firmware Versions 
    OS: 1.8.0.0
    Mono: 1.8.0.0
    Coprocessor: 1.8.0.0
    Protocol: 7
```

alexischr, 2024-04-30:
@duduita let's revisit this after we finish landing these recent network fixes.

NevynUK, 2024-04-30:
This needs retesting with the latest OS.

duduita, 2024-04-30:
I'm not seeing it anymore on 1.11.0.0 and 1.11.0.2, so after the 1.11.0.0 OS release, I'd say that we can close it :)

ctacke, 2024-05-01:
Fixed in 1.11