WildernessLabs/Meadow_Issues #433: SPI bus is dead after sleep/wake cycle

ctacke, 2024-01-23:
I have an app that runs a display over SPI.  All is good before sleep/wake.  After wake app code keeps running, but the display stops updating.  Putting a scope on the clock and data lines shows no activity after wake, but the app is calling to send SPI data.

ctacke, 2024-01-29:
Additional info:
Networking is still live after wake, so the SPI channel to the ESP is still fine.  The only difference that comes to mind is app SPI commands route through the UPD driver.

Note that the UPD driver itself becomes invalid (bad handle) after sleep/wake and there is code to re-open it on wake.

One other behavior note, on wake the SCK line is solid high.  MOSI is low.

Tests done:
- set the SPI bus handles to NULL on wake so that subsequent calls re-initialize the bus.  No effect.
- added "deinintialize" to the lower SPI driver, and called that on wake.  Only change is that SCK goes low.  Still no clocks or data

NevynUK, 2024-02-01:
Running the board with CoolTerm connected to the port normally used for USB gives the following output:

```
Power port state change
Powering off
x=243 direction=-1
ioctl GetSetConfigurationValue failed 0
UnhandledException System.InvalidOperationException: Cannot apply a context that has been marshaled across AppDomains, that was not acquired through a Capture operation or that has already been the argument to a Set call.
Wake on interrupt
Powering on
x=213 direction=-1
 at System.Threading.ExecutionContext.Run (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x00025] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at Meadow.App`1[D].InvokeOnMainThread (SyÇstem.Action`1[T] action, System.Object state) [0x0002a] in <6d54efa3047c4b3994f8919d585c22f1>:0 
  at Meadow.Foundation.Graphics
icroLayout.DisplayScreen.DrawLoopThreaded () [0x00009] in <2d04c050e59046da8835edb082a49068>:0 
  at Meadow.Foundation.Graphics.MicroLayout.DisplayScreen.DrawLoop () [0x0000e] in <2d04c050e59046da8835edb082a49068>:0 
  at System.Threading.ThreadHelper.ThÇreadStart_Context (System.Object state) [0x00014] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.ExecutionContext
RunInternal (System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00071] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.ExecutionContext.Run. 
System.Threading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state, System.Boolean preserveSyncCtx) [0x00000] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.ExecutionContext.Run (System.ThreÇading.ExecutionContext executionContext, System.Threading.ContextCallback callback, System.Object state) [0x0002b] in <e7054f2a5c
94bb7bb05b4a0fedd0bbb>:0 
  at System.Threading.ThreadHelper.ThreadStart () [0x00008] in <e7054f2a5c394bb7bb05b4a0fedd0bbb>:0 
x=183 direction=-1
```

Looks like SPI may be OK and the problem is elsewhere.

bryancostanich, 2024-02-01:
Thanks @NevynUK for sleuthing and @ctacke for picking it up and fixing it.

closed 2024-02-27 by bryancostanich