WildernessLabs/Meadow_Issues #434: Driver for UBLOX SARA R4 Modem

alaindlk, 2024-01-24:
Hello I've previously used a sparkfun SARA R4 shield using the UBLOX LTE Modem.

The modem works on 1.8V TTL but the sparkfun shield includes two level shifters feeded by the 5V rail.
It uses Pin 6 and 7 on the shield to Reset (10 sec LOW Value) and Start (3 SEC LOW value) the modem.
The UART speedport has to be manually configured using the AT command AT+IPR command. On restart it uses the previously set speed .

The remainder seems to be the traditional At commands used in cellular modems. You'll find in attachement the AT command file used in the sara R4
[SARA-R4___SARA-N4_AT_Commands_Manual.pdf](https://github.com/WildernessLabs/Meadow_Issues/files/14041075/SARA-R4___SARA-N4_AT_Commands_Manual.pdf)

bryancostanich, 2024-01-24:
@alaindlk thanks for the driver request! We're clearing out the last of our priority-0 bugs, but we'll get on this right after. Looks like [this is the modem](https://www.sparkfun.com/products/14997) board yeah? We'll get them purchased ASAP. 

In the meantime, if you need immediate cellular support, I recommend picking up one of the modems listed on our [cellular guide](https://developer.wildernesslabs.co/Meadow/Meadow.OS/Cellular/).

Thanks!

alaindlk, 2024-01-24:
@bryancostanich Yes that's the board I'm using...