WildernessLabs/Meadow_Issues #435: Network generating a wake interrupt? Sort of?

ctacke, 2024-01-25:
Odd bug, but very, very repeatable.

I have a test app (Project Lab `WakeOnInterrupt` sample).  It sleeps on a falling interrupt, wakes on rising on the same pin.  Sleep wake works very reliably with this sample (except the SPI not coming back, but that's a separate issue).

I extended the sample to connect to the network and then continually call postman echo.  And I get the following behavior:

- power up
- network connects
- postman returns data
- falling interrupt
- device goes to sleep
- rising interrupt
- device wakes
- postman returns data
- falling interrupt
- device goes to sleep
- device "half wakes" best I can describe - the USB connection on the PC sees it, but comes up as "Unknown USB Device - Device Descriptor Request failed"

At this point 
- no interrupts do anything - device won't wake or sleep
- device is not enumerating as a serial device, so can't check for output
- 99% sure code is not running as the GPIO toggle for the display backlight is not changing state

ctacke, 2024-01-25:
Test App:
[NetworkWake.zip](https://github.com/WildernessLabs/Meadow_Issues/files/14056221/NetworkWake.zip)

ctacke, 2024-10-15:
@Peter-Moody have you validated/tested this since all of the sleep work you completed?

Peter-Moody, 2024-10-15:
No

halyssonJr, 2025-01-16:
In this sample I changed the external interrupt, instead of pin A0 I used **mikroBus** INT, but I kept the same idea. Well, there are some improvements related to wake/up, but a specific point I noticed was when I pressed the button during a long time and started a request , the HTTP Task, was blocked and stopped the request. Also, I shared a log showing exactly what happened. Regarding to the backlight I suspected this pin is connected directly to Vcc, but I could see the animation stopping and continue when wake up.

https://github.com/user-attachments/assets/62d8a7d5-a5ef-477d-9916-9626c4e88bfa

[log.txt](https://github.com/user-attachments/files/18442780/log.txt)