WildernessLabs/Meadow_Issues #437: FT232H driver - I2C Exception when GPIO Set High

E-R-T-W, 2024-01-29:
**Describe the bug**
When using FT232H GPIO independently of I2C, all is well.
When using FT232H I2C independently of GPIO, all is well.
But when using the 2 together, as soon as a GPIO is set high, I2C crashes with an unrecoverable exception, whether reading I2C direct or using the sensor.StartUpdating() and sensor.Updated() event.

**To Reproduce**
Steps to reproduce the behavior:
Run the following code, commenting out either I2C or GPIO and all works fine, but when run both together (and comment out one of the I2C direct read or event based read) and as soon as a GPIO is set high, I2C crashes with an unrecoverable exception.
`using Meadow;
using Meadow.Foundation.ICs.ADC;
using Meadow.Foundation.ICs.IOExpanders;
using Meadow.Hardware;
using Meadow.Units;

namespace I2C_IO
{
    public class MeadowApp : App<Windows>
    {
        private Ft232h _expander = new Ft232h();
        private Ads1115 _ads1115;
        private II2cBus _bus;

        private IDigitalOutputPort _c5;

        public static async Task Main(string[] args)
        {
            await MeadowOS.Start(args);
        }

        public override Task Initialize()
        {
            Resolver.Log.Info("Initialize...");

            try
            {
                _bus = _expander.CreateI2cBus(0, I2cBusSpeed.Standard);
                _ads1115 = new Ads1115(_bus,
                    Ads1x15Base.Addresses.Address_0x48,
                    Ads1x15Base.MeasureMode.OneShot,
                    Ads1x15Base.ChannelSetting.A0SingleEnded,
                    Ads1115.SampleRateSetting.Sps128);
                //_ads1115.Updated += _ads1115_AnalogUpdated;
                //_ads1115.StartUpdating(TimeSpan.FromMilliseconds(1000));

                _c5 = _expander.CreateDigitalOutputPort(_expander.Pins.C5);
            }
            catch (Exception ex)
            {
                Resolver.Log.Info($"Error in Initialize(): {ex}");
            }

            return base.Initialize();
        }

        private void _ads1115_AnalogUpdated(object? sender, IChangeResult<Voltage> e)
        {
            try
            {
                Resolver.Log.Info($"A0 = {e.New.Volts.ToString("0.000")}");
            }
            catch (Exception ex) 
            {
                Resolver.Log.Info($"Error in _ads1115_AnalogUpdated(): {ex}");
            }
        }

        public override async Task Run()
        {
            Resolver.Log.Info("Run...");

            int i = 0;
            while (true)
            {
                try
                {
                    Resolver.Log.Info($"A0_Read_Volts = {(await _ads1115.Read()).Volts.ToString("0.000")}");
                    Resolver.Log.Info($"A0_ReadRaw = {await _ads1115.ReadRaw()}");
                    if (i % 5 == 0 && i > 0)
                    {
                        _c5.State = !_c5.State;
                    }
                    Resolver.Log.Info($"C5 = {_c5.State}");
                }
                catch (Exception ex)
                {
                    Resolver.Log.Info($"Error in Run(): {ex}");
                }

                i++;
                await Task.Delay(1000);
            }
        }
    }
}`

**Expected behavior**
FT232H driver should be able to do I2C comms and GPIO at the same time.

**Screenshots**
`Initializing OS...
Using default app.config.yaml...
Log level: Information
Update Service is disabled.
Health Metrics disabled.
Initialize...
Run...
A0_Read_Volts = 1.005
A0_ReadRaw = 16077
C5 = False
A0_Read_Volts = 1.005
A0_ReadRaw = 16077
C5 = False
A0_Read_Volts = 1.005
A0_ReadRaw = 16077
C5 = False
A0_Read_Volts = 1.005
A0_ReadRaw = 16078
C5 = False
A0_Read_Volts = 1.005
A0_ReadRaw = 16077
C5 = False
A0_Read_Volts = 1.005
A0_ReadRaw = 16077
C5 = True
Error in Run(): System.IO.IOException: Error writing device while setting up address
   at Meadow.Foundation.ICs.IOExpanders.Ft23xxI2cBus.Write(Byte peripheralAddress, Span`1 writeBuffer)
   at Meadow.Foundation.I2cCommunications.Write(Span`1 data)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.SetRegister(Register register, UInt16 value)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.SetConfigRegister(UInt16 value)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.ReadRaw()
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.ReadSensor()
   at I2C_IO.MeadowApp.Run() in D:\Data_Weyco\VS2022\Meadow\Meadow.Desktop.Samples - Copy\Source\Windows\I2C+IO\I2C+IO\MeadowApp.cs:line 68
Error in Run(): System.IO.IOException: Error writing device while setting up address
   at Meadow.Foundation.ICs.IOExpanders.Ft23xxI2cBus.Write(Byte peripheralAddress, Span`1 writeBuffer)
   at Meadow.Foundation.I2cCommunications.Write(Span`1 data)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.SetRegister(Register register, UInt16 value)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.SetConfigRegister(UInt16 value)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.ReadRaw()
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.ReadSensor()
   at I2C_IO.MeadowApp.Run() in D:\Data_Weyco\VS2022\Meadow\Meadow.Desktop.Samples - Copy\Source\Windows\I2C+IO\I2C+IO\MeadowApp.cs:line 68
Error in Run(): System.IO.IOException: Error writing device while setting up address
   at Meadow.Foundation.ICs.IOExpanders.Ft23xxI2cBus.Write(Byte peripheralAddress, Span`1 writeBuffer)
   at Meadow.Foundation.I2cCommunications.Write(Span`1 data)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.SetRegister(Register register, UInt16 value)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.SetConfigRegister(UInt16 value)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.ReadRaw()
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.ReadSensor()
   at I2C_IO.MeadowApp.Run() in D:\Data_Weyco\VS2022\Meadow\Meadow.Desktop.Samples - Copy\Source\Windows\I2C+IO\I2C+IO\MeadowApp.cs:line 68`

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Win11.23H2 Pro
 - IDE and version: Visual Studio Professional 2022 v17.8.5
 - Meadow extension for IDE version: 1.5.0

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: Meadow.Windows 1.8.0.1
 - Meadow OS version: Meadow.Windows 1.8.0.1
 - If they are different, please provide these versions as well.
   - Meadow Mono version: N/A (Meadow.Windows)
   - Meadow coprocessor version: N/A (Meadow.Windows)

**Additional context**
Issue discussed on Slack at:
https://wildernesslabspublic.slack.com/archives/CQWD660E8/p1706489627714189?thread_ts=1706478526.963999&cid=CQWD660E8

engunneer, 2024-01-29:
I suspect this is a limitation of the workaround they used to get I2C working at all. To use I2C, they are working around an apparent bug in libmpsse.dll by using an HID driver. The GPIO commands are using the MPSSE command set, so probably suffers from the same issues as trying to use SPI at the same time as I2C. See the [Ft232 Readme](https://github.com/engunneer/Meadow.Foundation/blob/develop/Source/Meadow.Foundation.Peripherals/ICs.IOExpanders.Ft232h/readme.md#hid-for-i2c) for more info.
Before the HID driver option came out, I spent a fair bit of time trying to debug libmpsse.dll, and was able to reproduce, but not solve the I2C issue. I also looked at doing a full rewrite of the Ft232h Driver based on an nuget package that wraps some FTDI code in .NET, and have a working example Winforms app with "proper" I2C support to refence. It's a bit of a monster to extract into it's own library, but it should be possible. This could also unlock using multi-MPSSE chips like the FT2232.

ctacke, 2024-01-30:
I ran a test with I2C and GPIO and I see I2C data coming it, GPIO toggling and no exception with this code:

```
var ft232 = new Ft232h();

await TestI2CWithGPIO(ft232);

async Task TestI2CWithGPIO(Ft232h expander)
{
    var sensor = new Pct2075(expander.CreateI2cBus());
    var output = expander.CreateDigitalOutputPort(expander.Pins.C0);

    bool state = true;

    while (true)
    {
        var reading = await sensor.Read();
        Debug.WriteLine($"Temp: {reading.Fahrenheit}F");
        await Task.Delay(1000);
        output.State = state;
        state = !state;
    }

}
```

ctacke, 2024-01-30:
I'll swap peripherals - it might be I2C write and GPIO colliding. Just as a point of info, the exception you're getting is definitely a NACK at the target address, and I can easily repro the error by disconnecting one of the I2C bus lines.

E-R-T-W, 2024-01-31:
That's interesting. I took your code and made a small tweak to see if setting GPIO low again would make a difference. It doesn't.
`            while (true)
            {
                var reading = await sensor.Read();
                Debug.WriteLine($"Voltage: {reading.Volts}V");
                await Task.Delay(1000);
                output.State = true;
                await Task.Delay(1000);
                output.State = false;
                await Task.Delay(1000);
            }`
The first read works.
Setting the GPIO high then low works.
But the second read crashes with the exception:
`App Error in App Initialize: Error writing device while setting up address
 System.IO.IOException: Error writing device while setting up address
   at Meadow.Foundation.ICs.IOExpanders.Ft23xxI2cBus.Write(Byte peripheralAddress, Span`1 writeBuffer)
   at Meadow.Foundation.I2cCommunications.Write(Span`1 data)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.SetRegister(Register register, UInt16 value)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.SetConfigRegister(UInt16 value)
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.ReadRaw()
   at Meadow.Foundation.ICs.ADC.Ads1x15Base.ReadSensor()
   at I2C_IO.MeadowApp.TestI2CWithGPIO(Ft232h expander) in D:\Data_Weyco\VS2022\Meadow\Meadow.Desktop.Samples - Copy\Source\Windows\I2C+IO\I2C+IO\MeadowApp.cs:line 65
   at I2C_IO.MeadowApp.Initialize() in D:\Data_Weyco\VS2022\Meadow\Meadow.Desktop.Samples - Copy\Source\Windows\I2C+IO\I2C+IO\MeadowApp.cs:line 29
   at Meadow.MeadowOS.Start(String[] args, IApp app)`
   
I'm thinking maybe the error is specific to the ADS1115 chip and its driver.
I do have a second ADS1115 I tested and the same thing happens.

Just to play with, I ordered a BME680 and it came in. I'll hook it up tonight (hopefully if I have time) and test it and let you know the results.

E-R-T-W, 2024-01-31:
Not sure why the line breaks are not coming through - kind of annoying.
Sorry for the mess with formatting.

E-R-T-W, 2024-02-02:
I had chance to test the BME680 tonight.
Same problem.
There's definitely a bug with using I2C concurrently with GPIO!

ctacke, 2024-03-26:
This is fixed with the FT232 driver that shipped in 1.9

closed 2024-03-26 by ctacke