WildernessLabs/Meadow_Issues #439: Sockets TcpListener and TcpClient - ESP32 "disappears" - Can't even ping

E-R-T-W, 2024-01-29:
**Describe the bug**
wrt Sockets (TcpListener and TcpClient) and follow-up to the resolved lockstep issue [#347](https://github.com/WildernessLabs/Meadow_Issues/issues/347)...
The maximum number of threads on the ESP32 is 8 -> 1 for the server and 7 for concurrent clients. This maximum number of clients has to be self managed in code i.e. if you just keep trying to connect clients to the Meadow acting as a server, you'll crash it.

My first attempt at self managing the number of clients was to accept 6 clients as normal. Upon the 7th client connection, after establishing the connection, I would have the Meadow server inform the client that the connection was being rejected and drop it.
There's a Meadow bug with that approach that seems to perhaps lie in object disposal??? Within several minutes to a few hours, the ESP32 simply "disappears" i.e. you can't even ping it anymore and all comms with the Meadow drop, but the Meadow continues to run (as evidence by the blinking LED still functioning). There are no errors of any kind, and a watchdog will not guard against this problem either. Simply comms stop and you can no longer even ping the ESP32.

The solution has been to not even accept more than 7 connections in the first place. It's a better solution in some regards, but it lacks the ability to inform the client that it is being dropped due to too many concurrent client connections. Nonetheless, there's a Meadow bug that seems to have to do with disposing objects that causes the ESP32 to "disappear" (no errors, Meadow still runs, but comms are down, watchdog cannot guard against this bug, the WiFi events of error and/or loss in WiFi don't fire either - you simply can't even ping the ESP32 - makes troubleshooting the problem quite problematic).

I've attached the source code, which is very similar, for both versions of the Meadow server for your review and debugging.
The client code is identical in both scenarios.

**To Reproduce**
Run the Listener 1.0.0.1 (with bug) source code on the Meadow, and then run multiple instances of the Client 1.0.0.1.
Start a continuous ping to the Meadow.
Anywhere from several minutes to several hours, the ESP32 will "disappear" and be non pingable, but the Meadow continues to run without error.

In contrast, Run the Listener 1.0.0.1 without the bug on the Meadow, and then run multiple instances of the Client 1.0.0.1.
This works flawlessly for days on end, connecting and disconnecting as many clients as you want.
If you run many many clients and close the connected clients, the other running clients will establish connection and work perfectly well.

It should not matter which server version of the source code is run, the ESP32 should not bomb!

**Expected behaviour**
You shouldn't even have to self manage the number of concurrent client connections, however, even if you do self manage the number of concurrent client connections, the ESP32 should never "disappear" and be non pingable with zero errors (the WiFi events don't even fire for network disconnected or error), while the Meadow continues to run.

It should not matter which server version of the source code is run, the ESP32 should not bomb!

**Screenshots**
No screenshots required - run the server and many multiple clients, along with a continuous ping and you will see the ESP32 bombs.

**Developer tools (please complete the following information as best as you can):**
 - OS and version: Win11.23H2 Pro
 - IDE and version: Visual Studio Professional 2022 v17.8.5
 - Meadow extension for IDE version: 1.5.0

**Meadow (please complete the following information as best as you can):**
 - Meadow hardware version: F7FeatherV2
 - Meadow OS version: 1.8.0.0
 - Meadow Mono version: 1.8.0.0
 - Meadow coprocessor version: 1.8.0.0

**Additional context**
 - See the resolved issue [#347](https://github.com/WildernessLabs/Meadow_Issues/issues/347) wrt max ESP32 allowed threads of 8: 1 server and 7 concurrent clients allowed.
 - Source Code Attached:

[Listener 1.0.0.1 (with bug).zip](https://github.com/WildernessLabs/Meadow_Issues/files/14090442/Listener.1.0.0.1.with.bug.zip)

[Listener 1.0.0.1.zip](https://github.com/WildernessLabs/Meadow_Issues/files/14090466/Listener.1.0.0.1.zip)

[Client 1.0.0.1.zip](https://github.com/WildernessLabs/Meadow_Issues/files/14090478/Client.1.0.0.1.zip)

E-R-T-W, 2024-01-29:
It could be my imagination, but if you occasionally and randomly close connected client consoles and open new ones, the problem appears to occur quicker.
Sometimes you have to wait over an hour or two for the bug to rear it's head.

NevynUK, 2024-02-09:
> Run the Listener 1.0.0.1 (with bug) source code on the Meadow, and then run multiple instances of the Client 1.0.0.1.

@E-R-T-W How many clients were you able to run ?

E-R-T-W, 2024-02-09:
@NevynUK With the example code I provided, 2 clients successfully connect and communicate, echoing an incrementing counter. The 3rd client connects, but the server informs the client that the 3rd connection is being rejected and drops it. The 3rd client then tries to connect again and the server does connect, but informs the client again that the connection is being refused due to too many connections and the server drops the 3rd connection, and so on, and so on.

In my test, I ran like 20 clients (but it doesn't matter how many - any number beyond 2 where connections are being dropped causes the bug). 2 always connected successfully and communicated, echoing an incrementing counter.
"Randomly", 1 of the other 18 clients would connect, but then be informed their connection was being dropped. The other 18 clients would continually try to connect, and "randomly" 1 of them would make a successful connection but the server would always drop the 3rd connection because only 2 connections were allowed.

Does that make sense or do I need to provide further explanation, example?

The code all works exactly as it should - the only problem is eventually (from minutes to a few hours), the ESP "disappears" and you can't even ping it anymore, but there are no errors of any kind. It's like there's a memory leak, an issue/bug with disposing objects when dropping all the connections.

In my real world scenario, I have 150+ HMI's trying to read the F7s and so my F7 code has to manage the number of connections in order not to crash the F7, since the ESP32 can only handle 8 threads and 1 of those is reserved for STM to ESP32 comms.

NevynUK, 2024-02-10:
Your explanation makes sense, I had missed the fact that the connection being dropped for clients 3 and above was necessary to reproduce the issue.

So the good news is I have managed to reproduce the issue.

The bad news, apart from the fact I have reproduced the issue is that it took over 36 hours to reproduce.

The ESP32 is restarting, I ca see that from the debug output.  It looks like a clean restart though, there is no traceback information in the serial output.  Normally failed code on the ESP32 gives trace of the methods being called if the issue is on the ESP32.

To be honest, I was hoping it was in the ESP32 as these issues tend to be easier to fix.

On to look at what the STM32 is doing...

Back in 2 days :)

E-R-T-W, 2024-02-10:
I figured this one would be a super difficult one to track down, partially because it takes so long for the bug to occur. I've never been able to run it as long as you did though, usually max 2 to 4 hours, and sometimes less than 1.
I'm curious, did you just leave it run on it's own?

Like I said in one of the comments, it could be my imagination, but I've done a heck of a lot of testing on this one and it seems to occur quicker if you shutdown some of the clients and start new ones, including the connected ones and let some of the rejected ones connect.

NevynUK, 2024-02-10:
Yes, I just left it running on my dev box overnight.

Going to set up a separate box to run this unattended.

NevynUK, 2024-02-14:
Made a little progress.  Run for just over 48 hours but failed around 36 hours.

```
poll_event_wait: mono_poll () failed, error (115) Network is dow  n
Wifi_NetworkDisconnected event fired
Calling Device.PlatformOS.Reset() since only way to re-establish WiFi connectivity!!
```

Followed by:

```
Error connecting to access point: Connection failed: CannotConnectToAccessPoint
Wifi_NetworkDisconnected event fired
Calling Device.PlatformOS.Reset() since only way to re-establish WiFi connectivity!!!
```

The third attempt to reboot ended with a successful connection and the application on the device and the client applications on the desktop started running again.

Now to find out why `mono_poll` is failing.

E-R-T-W, 2024-02-15:
Interesting – I was never able to get those events to fire for this scenario.

It’ll be nice to have the bug fixed, but in the short term, if those events would consistently fire, at least there’d be a sort of poor work around by resetting the device.

NevynUK, 2024-02-16:
@E-R-T-W Are you on Slack?  If you are can you DM me (I'm mark on the public slack) please, might be easier to discuss the difference we are seeing that way - time zones permitting :)

E-R-T-W, 2024-02-22:
Yes I am on Slack, but under a different name.
I'm on vacation this week (no WL hardware with me), but will DM you next week when I'm back home.

NevynUK, 2024-03-07:
Discussions on the issue ongoing in Slack.

Agreed we can drop this to P1.

ctacke, 2025-03-18:
@NevynUK did this get fixed?

alexischr, 2025-03-18:
I will re-test - we may have problems with socket exhaustion cases