WildernessLabs/Meadow_Issues #440: Documentation issue BME688 sensor

NevynUK, 2024-02-01:
There are a number of missing images in the BME688 sensor driver:

https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Atmospheric.Bme688.html

```
To wire a Bme688 to your Meadow board, connect the following:

<img src="../../API_Assets/Meadow.Foundation.Sensors.Atmospheric.Bme688/Bme688_Fritzing.png"
```

patridge, 2024-02-02:
Here's the API docs include file and the listed missing image.
https://github.com/WildernessLabs/Documentation/blame/main/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Atmospheric.Bme688.md#L118

I don't see any other images that are missing, but there is currently an issue with this API docs page not getting the include. It's possible there were multiple broken images as a result of WildernessLabs/Documentation#504. You can still see the old version with the include in a limited form [via Bing cache](https://cc.bingj.com/cache.aspx?q=https%3a%2f%2fdeveloper.wildernesslabs.co%2fdocs%2fapi%2fMeadow.Foundation%2fMeadow.Foundation.Sensors.Atmospheric%2fBme688%2f&d=4976024900484324&mkt=en-US&setlang=en-US&w=KPJh_guXllbW1fOB-Ko67lqRc6Ii_x8g). So, there may only be the one image needing a fix.

As for the referenced image, I am not finding any such BME688 Fritzing diagram to use. There is only a [similar one for BME680](https://github.com/WildernessLabs/Documentation/blob/main/docs/API_Assets/Meadow.Foundation.Sensors.Atmospheric.Bme680/Bme680_Fritzing.png).

patridge, 2024-02-02:
Looks like there is another quirk of that new Docusaurus site. The include is being included here (but not [here](https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Atmospheric/Bme688/), note the `/` between the class name and the namespace).
https://developer.wildernesslabs.co/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Atmospheric.Bme688/

alexischr, 2024-02-06:
Just checked real quick and this is still broken

closed 2024-02-06 by patridge

patridge, 2024-02-06:
This has been addressed on the dev release of docs.
https://developer.wildernesslabs.dev/docs/api/Meadow.Foundation/Meadow.Foundation.Sensors.Atmospheric.Bme688/#wiring-example
We'll want to close it when those are published to prod.