WildernessLabs/Meadow_Issues #443: CLI V2 Timeout writing firmware

NevynUK, 2024-02-05:
I've had this a few times now writing firmware to a board (2.c)

Command `meadow firmware write`

```
Writing all firmware for version '1.8.0.0'...
Flashing OS with /Users/markstevens/.local/share/WildernessLabs/Firmware/1.8.0.0/Meadow.OS.bin
dfu-util 0.11

<cut...>

Transferring bootloader.bin to coprocessor...
File received, flashing ESP32
Transferring partition-table.bin to coprocessor...
File received, flashing ESP32100%

The operation has timed out.
```

NevynUK, 2024-02-05:
I'm seeing this with the following as well:

* runtime enable
* runtime disable

These should finish very quickly so a time out of 30s for these two commands is way too long.

The `firmware write` command also finishes doing its work but then times out.

These commands all have one thing in common - they reboot the board at the end of the command.

alexischr, 2024-02-06:
There is a workaround - trying again. Setting this as `p1`

NevynUK, 2024-02-07:
Trying again is not a work around as it does not work.  You still get the timeout on the second, third..... attempt.

adrianstevens, 2024-02-18:
@NevynUK is this still an issue? (hoping it was fixed with the recent serial loop fix)

NevynUK, 2024-02-21:
@adrianstevens I have not been able to test the full OS/RT/ESP deploy as a single command as the OS deploy is not working at the moment.

adrianstevens, 2024-03-07:
This should be fixed now

closed 2024-03-26 by adrianstevens