WildernessLabs/Meadow_Issues #444: CLI V2 does not show application file deployment

NevynUK, 2024-02-05:
Steps:

* Deploy an app to the board and normal
* Disable the application `meadow runtime disable`
* Now issue the command `meadow file delete App.dll` We should now have the full application without the `App.dll` file on the board
* Issue the command `meadow app deploy` to write the `App.dll` file to the board. 

You see the following output:

```
Generating list of files to deploy...
Deleting file 'meadow.log'...
Deleting file 'dns.conf'...
Deleting file 'Cache'...
Deleting file 'Data'...
Deleting file 'Documents'...
Deleting file 'Temp'...
Deleting file 'update-store'...
```

There is no mention of the `App.dll` file that has actually be written to the board successfully.  This leaves you with the impression that nothing has been deployed to the board.

KallDrexx, 2024-02-06:
Note that the `App.dll` deploy does show up, but once it gets to `100%` the line gets removed

NevynUK, 2024-02-07:
I think this should be higher than P2.  From a user perspective I need feedback that the command has worked or that it has not.

In the current state I get feedback that something else (the file deletion) has happened and not that the command I issued has worked correctly.

adrianstevens, 2024-02-23:
I'll test on Mac

adrianstevens, 2024-02-23:
fixed in v2.0.10.0

adrianstevens, 2024-03-07:
Fixed - closing issue

closed 2024-03-07 by adrianstevens