WildernessLabs/Meadow_Issues #446: ProjectLab3 has no input after deploy until reset

KallDrexx, 2024-02-07:
**Describe the bug**
When deploying a new application to my ProjectLab3, the PL3's input buttons are not responsive until the meadow has been restarted

**To Reproduce**
Steps to reproduce the behavior:

1. Create a project with the following code

```csharp
 public class MeadowApp : App<F7CoreComputeV2>
{
    public override async Task Run()
    {
        Console.WriteLine("Starting App");

        var projectLab = ProjectLab.Create();
        var button = projectLab.UpButton!;
        button.PressEnded += (s, e) => Console.WriteLine("Button Pressed");

        Console.WriteLine("Waiting for up button press");
        while (true)
        {
            await Task.Delay(10);
        }
    }
}
```

2. Deploy with `dotnet build -c Release && meadow app deploy .\bin\Release\netstandard2.1\App.dll && meadow runtime enable && meadow listen`
3. Wait for deploy to complete and the `Waiting for up button press`
4. Press the up button
---> Observe the button never responds

5. Press the reset button
3. Wait for deploy to complete and the `Waiting for up button press`
4. Press the up button
---> Observe we now see text stating the button was pressed

**Expected behavior**
A restart should not be requierd

**Developer tools (please complete the following information as best as you can):**
 - OS and version: [e.g., Windows 10 v22H2, macOS Monterey v12.6] Windows 11
 - IDE and version: [e.g., Visual Studio Professional 2022 v17.3.5, Visual Studio 2022 for Mac v17.3.5] N/A
 - Meadow extension for IDE version: [e.g., VS 2022 Tools for Meadow v0.19.9] N/A

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: [e.g. F7v2, CCMv2] CCMv2
 - Meadow OS version: [e.g., v0.6.7.19] 1.8.1.16
 - If they are different, please provide these versions as well.
   - Meadow Mono version: [e.g., v0.6.7.19]
   - Meadow coprocessor version: [e.g., v0.6.7.19] 1.8.0.0

KallDrexx, 2024-05-26:
As another point of data, this is not ProjectLab library related, as this can be reproduced with the following code:

```csharp
            var i2c = Device.CreateI2cBus();
            var interruptPin = Device.GetPin("A05") ?? throw new NullReferenceException("interruptPin");
            var resetPin = Device.GetPin("D05") ?? throw new NullReferenceException("resetPin");
            var interrupt = interruptPin.CreateDigitalInterruptPort(InterruptMode.EdgeRising, ResistorMode.InternalPullDown);
            var reset = resetPin.CreateDigitalOutputPort();
            var mcp = new Mcp23008(i2c, 0x20, interrupt, reset);
            var upPin = mcp.GetPin("GP0") ?? throw new NullReferenceException("upPin");
            var upInterrupt = upPin.CreateDigitalInterruptPort(InterruptMode.EdgeBoth, ResistorMode.InternalPullUp);
            upInterrupt.Changed += (sender, args) =>
            {
                Console.WriteLine($"Interrupt on pin {upPin.Name} state changed to {args.New.State}");
            };

            while (true)
            {
                await Task.Delay(100);
            }
```

adrianstevens, 2024-05-26:
So your workflow won't run the linker, it's not necessarily an issue (unless you run out of space), but we typically validate against linked binaries. You should add `meadow trim` or just use the `meadow run` command that will build, link, deploy and listen.

I'm just testing now to see if I can repro.

adrianstevens, 2024-05-26:
And I am testing using the latest code in the develop branches and the latest OS (v1.2.0.0)

adrianstevens, 2024-05-26:
I ran into an issue with OS v1.2.0.0 so moved back to v1.11.3.2

I've been able to deploy and use buttons vis the VS2022 extension, the `meadow app run -c Release` command and the command listed above.

Testing on Project Lab v3.x.

Let's leave this issue open, if its consistent for you it's likely someone else will hit it.

adrianstevens, 2024-07-24:
I believe this ended up being a workflow issue - closing

closed 2024-07-24 by adrianstevens