WildernessLabs/Meadow_Issues #448: CLI V2 `meadow firmware write runtime`

NevynUK, 2024-02-07:
I have created a new firmware directory using the `1.8.1.0` as a template so I have OS etc in the directory.  This was renamed to `1.8.1.3` which is the version I have built locally.

Copied the OS and RT bins into the new directory.

Executed the command `meadow firmware write runtime -v 1.8.1.3` and received the following output:

`DFU is only used for OS files - select an OS file or remove the DFU option`

adrianstevens, 2024-02-18:
Fixed in v2.0.5

NevynUK, 2024-02-18:
> Fixed in v2.0.5

No it isn't fixed for Mac.  Just tested with 2.0.7.1 and the flash still times out.

Command: `meadow firmware write runtime -v 1.8.0.0`

Terminal output:

```
Getting runtime for 1.8.0.0...
Mono is disabled

Writing Runtime ...
Writing Meadow.OS.Runtime.bin: 10%
Writing Meadow.OS.Runtime.bin: 20%
Writing Meadow.OS.Runtime.bin: 30%
Writing Meadow.OS.Runtime.bin: 40%
Writing Meadow.OS.Runtime.bin: 50%
Writing Meadow.OS.Runtime.bin: 60%
Writing Meadow.OS.Runtime.bin: 70%
Writing Meadow.OS.Runtime.bin: 80%
Writing Meadow.OS.Runtime.bin: 90%
Writing Meadow.OS.Runtime.bin: 100%

The operation has timed out.
```

So attaching the OS debug console you see this:

```
[   45.837000] [ 8] (Info) file/hcom_file_dnld_stm32f7.c@121-Meadow downloading file (FileLen:3145728, Crc:0x859e819c, Name:/meadow0/update/os/Meadow.OS.Runtime.bin)
[  104.239000] [ 8] (Note) End of file write received
```

Now if you hit the reset button you see the following:

```
Meadow initialization has begun at 16:36:26 UTC Meadow time.

[    0.000000] [ 0] stm32_rng_initialize: Initializing RNG
[    0.079000] [ 3] meadow_upd_initialize
[    0.079000] [ 3] F/S creating 1 partition
[    0.079000] [ 3] Partitioning complete. Init partitions
[    0.079000] [ 3] F/S Part Init complete, mount & format
[    0.079000] [ 3] hcom_nx/create_fs/hcom_nx_fs.c@253-Mounted partition 0 as '/dev/little0p0' to '/meadow0' for type 'littlefs'
[    0.080000] [ 3] F/S creation complete
[    0.704000] [ 8] (Note) Meadow 1.8.1.7 rebooted, H/W:F7CoreComputeV2, Mono:Disabled, Trace level:0x7f, to:none (ramlog)
[    0.705000] [ 8] (Note) Mono version: 1.8.3.0
[    0.709000] [ 8] Flashing that file /meadow0/update/os/Meadow.OS.Runtime.bin
hcom_nx/diag/hcom_nx_trace_msg_proc.c@608-buffer corrupted or messages w/o linefeed. Deleted data.

hcom_nx/diag/hcom_nx_trace_msg_proc.c@608-buffer corrupted or messages w/o linefeed. Deleted data.

> Lots more corrupted messages deleted

hcom_nx/diag/hcom_nx_trace_msg_proc.c@608-buffer corrupted or messages w/o linefeed. Deleted data.

hcom_nx/diag/hcom_nx_trace_msg_proc.c@608-buffer corrupted or messages w/o linefeed. Deleted data.

Meadow initialization has begun at 16:37:39 UTC Meadow time.

[    0.000000] [ 0] stm32_rng_initialize: Initializing RNG
[    0.079000] [ 3] meadow_upd_initialize
[    0.079000] [ 3] F/S creating 1 partition
[    0.079000] [ 3] Partitioning complete. Init partitions
[    0.079000] [ 3] F/S Part Init complete, mount & format
[    0.079000] [ 3] hcom_nx/create_fs/hcom_nx_fs.c@253-Mounted partition 0 as '/dev/little0p0' to '/meadow0' for type 'littlefs'
[    0.080000] [ 3] F/S creation complete
[    0.707000] [ 8] (Note) Meadow 1.8.1.7 rebooted, H/W:F7CoreComputeV2, Mono:Disabled, Trace level:0x7f, to:none (ramlog)
[    0.708000] [ 8] (Note) Mono version: 1.8.0.0
[    0.718000] [ 8] (Warn) mono/hcom_mono_control.c@603-Mono will not start - version mismatch: Meadow.OS version 1.8.1.7, Mono version 1.8.0.0, Mono disabled
```

adrianstevens, 2024-02-21:
Additional work done - should be fully fixed in v20.8.0

NevynUK, 2024-02-21:
Tested 2.0.8.0 and it looks like this is fixed.

Thank you.

NevynUK, 2024-02-21:
Spoke too soon, so `meadow firmware write" works and does not time out.

`meadow firmware write runtime -f ./Meadow.OS.Runtime.bin` timesout.

adrianstevens, 2024-03-07:
working here on MacOS with CLI v2.0.20.0 ... @NevynUK  is this still an issue?

adrianstevens, 2024-03-07:
Tested with Meadow OS v1.9.1.8

adrianstevens, 2024-03-14:
fixed - closing issue

closed 2024-03-14 by adrianstevens