WildernessLabs/Meadow_Issues #451: PixelBufferBase should error on height or width of zero

KallDrexx, 2024-02-08:
**Describe the bug**
Currently, if you invoke `PixelBufferBase`'s constructor with a width or height of zero, no error occurs *until* you try and do something with the buffer (like `buffer.Clear()`).  Then an `IndexOutOfRangeException` occurs. 

This is confusing because it requires digging in a good bit to find out what is going on, and when width or height is a calculated value that's not expected to be zero (but is accidentally), then this can waste a bunch of debugging time.

**To Reproduce**
```csharp
var layerBuffer = new BufferRgb565(0, 25);
layerBuffer.Clear(Color.Black);
```

**Expected behavior**
Either creating a buffer with a zero height or width should throw an exception, or any operation that interacts with the buffer should have a more informative exception thrown.

adrianstevens, 2024-02-08:
easy fix - I'm on it

adrianstevens, 2024-02-22:
Fixed - will ship with v1.9.0.0

adrianstevens, 2024-03-07:
Shipped in v1.9.0 - closing issue

closed 2024-03-07 by adrianstevens