WildernessLabs/Meadow_Issues #454: MCP2515 driver for CAN

upswing1, 2024-02-09:
Driver for the MCP2515 stand-alone Controller Area Network (CAN) controller. It interfaces with microcontrollers (MCUs) through the Serial Peripheral Interface (SPI), enabling meadow to communicate in a CAN network until CAN support at the device level is implemented

engunneer, 2024-02-14:
I have two different MCP2515 based boards that have been on my "to play with someday" list. I may be able to look into this in the next few weeks. Do you have any particular thing you're trying to use it for?

engunneer, 2024-02-18:
@ctacke has a branch for MCP2515 on https://github.com/WildernessLabs/Meadow.Foundation/tree/feature/mcp2515. Do you have the ability to try it out?

adrianstevens, 2024-08-20:
complete - closing issue

closed 2024-08-20 by adrianstevens