WildernessLabs/Meadow_Issues #457: CLIv2 creates firmware folder before download failure

ctacke, 2024-02-14:
CLIv2 creates an empty folder for the firmware - for example firmware/1.8.0.0

If the download fails (e.g. auth failure) then the folder exists but is empty.

When you then attempt to download again, CLI says the firmware already exists locally (it looks like it's using the folder to test) but if you do a `firmware list` it doesn't show up.

Repro:

- Create an empty firmware folder with the name of the latest available firmware
- try `meadow firmware download` it will fail
- try `meadow firmware list` it will say the version is available

Workaround is to force: `meadow firmware download -f`

adrianstevens, 2024-02-15:
Since I've been playing in this code recently I'll take a look

adrianstevens, 2024-02-21:
this is fixed in v2.0.6.0

closing issue

closed 2024-02-21 by adrianstevens