WildernessLabs/Meadow_Issues #461: VS extension grabs serial port, even when not a meadow project

ctacke, 2024-02-14:
I have a meadow attached to my PC (Windows 11, VS2022)

- I run `meadow device info` from a console and get back info
- Open VS with a desktop project (no meadow projects in the solution)
- run `meadow device info` from the console 
```
> meadow device info
Getting device info...
Failed: Access to the path 'COM4' is denied.
```

CartBlanche, 2024-02-15:
@ctacke I just followed the repro steps you provided on my Windows 11 laptop, using VS2022 extension 1.8.1, with VS extension project loaded (so definitely not meadow) and tried the steps with both Meadow.CLI `v1.3.4` and then upgraded to `v2.0.5` and didn't hit this issue with either version of CLI.

adrianstevens, 2024-02-22:
I'm not able to repro either unfortunately

closed 2024-10-01 by alexischr