WildernessLabs/Meadow_Issues #467: CLIv2: Can only upload one file at a time?

alexischr, 2024-02-16:
**Describe the bug**
I should be able to write multiple files on the device using `meadow file write`
**To Reproduce**
#1:

```
for s in $(echo MeadowComms.bin bootloader.bin partition-table.bin); do meadow file write -f $s -t /meadow0/update/firmware/$s; done
Writing 1 file to device...
Writing 'MeadowComms.bin' as '/meadow0/update/firmware/MeadowComms.bin' to device
Timeout attempting to attach to device on /dev/tty.usbmodem3356317334381
^C^C^C^C
```

or  #2:

```
meadow file write -f MeadowComms.bin -t /meadow0/update/firmware/MeadowComms.bin -f MeadowComms.bin.md5 -t /meadow0/update/firmware/MeadowComms.bin.md5
Writing 2 files to device...
Writing 'MeadowComms.bin' as '/meadow0/update/firmware/MeadowComms.bin' to device
Writing 'MeadowComms.bin.md5' as '/meadow0/update/firmware/MeadowComms.bin.md5' to device
Error writing file: The operation has timed out.
```

**Expected behavior**
I should be able to script multiple file uploads, preferably in independent commands as in example #2

**Additional context**

`meadow v2.0.5`

alexischr, 2024-02-16:
Update: This seems to only be a problem when using the `-t` parameter.

adrianstevens, 2024-02-20:
@Peter-Moody  I'll try and reproduce but we may need to team up on this one

adrianstevens, 2024-02-25:
I was able to write 3 small files in a single command from Windows

adrianstevens, 2024-03-27:
@alexischr  is this still a problem for you?

alexischr, 2024-07-24:
The issue seems to be resolved, I tried the same commands several times:

```
for s in $(echo MeadowComms.bin bootloader.bin partition-table.bin); do meadow file write -f $s -t /meadow0/update/firmware/$s; done
Disabling runtime...
Writing 1 file to device...
Writing ‘MeadowComms.bin’ as ‘/meadow0/update/firmware/MeadowComms.bin’ to device
Writing  ‘MeadowComms.bin’: 100%
Writing 1 file to device...
Writing ‘bootloader.bin’ as ‘/meadow0/update/firmware/bootloader.bin’ to device
Writing  ‘bootloader.bin’: 100%
Writing 1 file to device...
Writing ‘partition-table.bin’ as ‘/meadow0/update/firmware/partition-table.bin’ to device
Writing  ‘partition-table.bin’: 100%
```

closed 2024-07-24 by alexischr