WildernessLabs/Meadow_Issues #468: Cannot use `meadow firmware write`

alexischr, 2024-02-16:
**Describe the bug**
I can no longer update my device using Meadow CLI. I am on Macbook Pro M2 MacOS 14.1.1.1.

**To Reproduce**

First I tried dfu-less OS updating, but the CLi instisted I had a dfu device:
```
meadow firmware write OS -d false
DFU device detected - using DFU to write OS
Flashing OS with /Users/lexas/Library/Application Support/WildernessLabs/Firmware/1.8.0.0/Meadow.OS.bin
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Warning: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release
dfu-util: No DFU capable USB device available
[null]
[null]
New meadow device not found
```

Then, I tried putting it into DFU myself and trying again. Same exact output as above (detected, then not detected)

Then, I used my own DFU command which worked fine (`dfu-util -a 0 -D Meadow.OS.bin -s 0x08000000`), so I used CLI for the Runtime upgrade:

```
meadow firmware write -f Runtime
/Users/lexas/.local/share/WildernessLabs/Firmware/1.8.0.0/Runtime
```

I just get some text, and no indication of error, or action.

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: CCMv2
 - Meadow OS version: 1.8.1.6

NevynUK, 2024-02-16:
To be fair, I have seen this `dfu-util: No DFU capable USB device available` and it is coming from `dfu-util` as I have also seen it when using `dfu-util` directly.

alexischr, 2024-02-16:
Yeah, but not 100% of the time (I used my DFU just fine on the first try), and it shouldn't fail when I didn't ask for DFU mode at all, as it is not the default anymore.

adrianstevens, 2024-02-16:
Interesting .... both of my Macs have been working 100% (i7 and M1 Pro)

adrianstevens, 2024-02-16:
@alexischr  are you using the latest CLI? v2.0.6?

alexischr, 2024-02-16:
I upgraded to 2.0.6 and have the same issue

adrianstevens, 2024-02-16:
I'm travelling this weekend so I won't be able to retest on a Mac until Monday unfortunately, but I'll prioritize this for next week ... hopefully we can track it down. Worst case we may need to hop on a call and debug on your machine.

adrianstevens, 2024-02-20:
ok dfu flashing is working reliably on my M1 Mac ... without dfu it times-out when writing the runtime on both Windows and Mac ... I think the CLI is getting ahead of the device

NevynUK, 2024-02-20:
I'm also seeing DFU issues on Mac M2.  Saw it this morning when testing new release.

RT timeout maybe related to issue 448.

I've had to go back to V1.

adrianstevens, 2024-02-20:
roger that - I'll try a few more devices and see if I can repro

adrianstevens, 2024-02-20:
good to know v1 works at least ... I'll compare the logic

NevynUK, 2024-02-21:
By default I get the same output as reported at the head of the issue.

If I manually put the device into DFU mode I get the following output:

```
Writing all firmware for version '1.8.3.0'...
Multiple devices found in bootloader mode - only connect one device
```

adrianstevens, 2024-02-25:
This may be fixed in the CLI v2.0.11-beta

adrianstevens, 2024-03-07:
Fixed and shipped - closing issue

closed 2024-03-07 by adrianstevens