WildernessLabs/Meadow_Issues #469: CLI V2 `meadow file list update` times out

NevynUK, 2024-02-16:
CLI v 2.0.6, Mac OS

Execute the command `meadow file list update` to get a list of the files in the `update` directory on the board.

This command times out.

Looking at the OS debug output we see the following:

```
[  371.203000] [ 8] (Error) file/hcom_file_dir_mgmt_utils.c@325-pathname 'update/' is invalid
[  371.203000] [ 8] (Error) file/hcom_file_dir_mgmt_utils.c@506-Eval pathname, ret:-22, errno:0
[  371.204000] [ 8] (Error) comms/hcom_host_preprocess.c@575-Init file list, ret:-22, errno:0
[  371.204000] [ 8] (Error) comms/hcom_host_preprocess.c@429-Processing data packet, ret:-22
```

So the OS cannot find the directory and is generating an error.  This error is being returned to CLI from the OS as:

```
hcom_host_send_simple_string_msg(HCOM_HOST_REQUEST_TEXT_ERROR, 0, "Error during download or delete command processing", thisFile, __LINE__);
```

If you change the command to `meadow file list /meadow0/update` then the command works.

adrianstevens, 2024-02-18:
Fixed in v2.0.7

NevynUK, 2024-02-18:
Confirmed fixed in 2.7.0.1.

closed 2024-02-21 by adrianstevens