WildernessLabs/Meadow_Issues #472: CLI V2 Firmware directory moves between V1 and V2

NevynUK, 2024-02-17:
The firmware directory on Mac has changed between .NET versions.  The latest CLI uses a different directory:

Old: `~/.local/share/WildernessLabs/Firmware/0.9.5.16/Meadow.OS.bin`
New: `~/Library/Application Support/WildernessLabs/Firmware/1.8.0.0/Meadow.OS.bin`

From Slack conversation:

Looks like a breaking change in .NET 8 with GetFolderPath and LocalApplicationData for Mac [GetFolderPath Changes](https://learn.microsoft.com/en-us/dotnet/core/compatibility/core-libraries/8.0/getfolderpath-unix)

"The recommended action in that article is to migrate files to the new directory and add a fallback check to the old directory."

adrianstevens, 2024-02-17:
thx @NevynUK !

adrianstevens, 2024-03-27:
Unless someone really wants this I don't think we'll bother adding logic to migrate

closed 2024-05-08 by adrianstevens