WildernessLabs/Meadow_Issues #473: Can't delete files in the Data folder via the CLI

adrianstevens, 2024-02-18:
Meadow OS 1.8.1.7 (the first one)
CLI v2.0.7

adrianstevens, 2024-02-20:
@Peter-Moody  is this by design?

Peter-Moody, 2024-02-20:
Yes, the HCOM additions to support subdirectories was, "just enough" for Chris to develop the workbench project he had in mind. One other goal was to that the HCOM changes should be backward compatible with CLIv1, which was all we really had at the time.

So there is no ability to create or remove subdirectories. They are auto created when the user want to add files to a new directory and whenever a subdirectory has the last file deleted the directory is deleted too. If the now empty directory is below other empty directories, all are deleted, upward until a directory is reached which contains files.

adrianstevens, 2024-02-22:
@Peter-Moody the issue I'm seeing is not being able to delete the files themselves in the data folder

The operation appears to compete but the files still show up when listing files. I don't see this issue with other folders.

adrianstevens, 2026-11-06:
Retested on CLI `meadow-3.0` (ccf6621) with a Project Lab (F7CoreComputeV2, OS 2.999.51.1, serial):

1. `meadow file write -f test.txt -t /meadow0/Data/cli473.txt`: written, and it reads back identical.
2. `meadow file delete /meadow0/Data/cli473.txt`: deleted.
3. `meadow file read /meadow0/Data/cli473.txt ...` afterwards: the device reports "could not be found".

Deleting files in `Data` works.

The device also removes a folder when its last file is deleted, so an empty `Data` folder disappeared too. The app recreates it at startup (`MeadowOS.InitializeFileSystem`).

closed 2026-11-06 by adrianstevens