WildernessLabs/Meadow_Issues #478: Need to capture time at the native level in ticks and surface through events

bryancostanich, 2024-02-21:
Right now we capture datetime in managed land when an interrupt comes in. this means that datetime is actually non-deterministic and always behind.

better if we captured the system.tickcount at native, passed that through to managed land, and then lazily evaluated datetime when necessary if we need duration between events.

Peter-Moody, 2025-09-30:
See issue #374 which is basically the same.

Peter-Moody, 2025-12-04:
I've done most of the work for the `Meadow.OS` side. By setting `MEADOW_INTERRUPT_INCLUDE_TIME_STAMP` to 1 in the `meadow_interrupt.h` file a subsequent `Meadow.OS` build will add the Nuttx tick count to the data sent to `Meadow.Core`.

Changing from the Nuttx tick to actual time (local or UTC) will be a simple task and some commented out sample code has been added to `meadow_interrupt.c`. 

Of course, `Meadow.Core` needs to be enhanced to handle this new timestamp information.