WildernessLabs/Meadow_Issues #480: CLI V2 `meadow firmware write -v 1.9.0.0` timeout (using update file NOT DFU)

NevynUK, 2024-02-21:
OS: Mac
CLI: 2.0.8.0

### Steps:

Flash a board with 1.8.3.0
Now execute the command `meadow firmware write -v 1.9.0.0`

### Output:
```
Writing all firmware for version '1.9.0.0'...
Mono is disabled
Writing Meadow.OS.Update.bin: 10%
Writing Meadow.OS.Update.bin: 20%
Writing Meadow.OS.Update.bin: 30%
Writing Meadow.OS.Update.bin: 40%
Writing Meadow.OS.Update.bin: 50%
Writing Meadow.OS.Update.bin: 60%
Writing Meadow.OS.Update.bin: 70%
Writing Meadow.OS.Update.bin: 80%
Writing Meadow.OS.Update.bin: 90%
Writing Meadow.OS.Update.bin: 100%

Getting runtime for 1.9.0.0...

Writing Runtime ...
Download of '/meadow0/update/os/Meadow.OS.Update.bin' success (checksums calculated:0xFDB9479C, expected:0xFDB9479C)
The operation has timed out.
```

So this time the system is attempting a full update using the `Meadow.OS.Update.bin` file and not DFU.

### Additional Information

So the board was left alone for over 2 minutes and there was no change in state of the board.

The board was then reset following the two minute delay and it then looked like the update process was taking place.  The board was left to its own devices for about 2-3 minutes at which point the board appeared to restart normally (with the RT disabled).  `meadow device info` confirmed that the board had updated the OS and RT.

adrianstevens, 2024-03-07:
Sounds like the same issue as #448 

Can you test with CLI v2.0.20.0?

adrianstevens, 2024-03-14:
@NevynUK any luck with the latest CLI release?

NevynUK, 2024-03-14:
Closing as this is working in 2.0.21

closed 2024-03-14 by NevynUK