WildernessLabs/Meadow_Issues #482: BME280 returns bad data

ctacke, 2024-02-21:
Simple test app on a CCM platform:

```
var bme = new Bme280(Device.CreateI2cBus());

while (true)
{
    Resolver.Log.Info("Reading...");
    var t = await bme.Read();
    Resolver.Log.Info($"{t.Temperature.Value.Fahrenheit}F");
    await Task.Delay(1000);
}
```

Output is:
```
Reading...
15.7099996566772F
Reading...
15.6919992446899F
Reading...
15.6919992446899F
```

I definitely see activity on the bus with a scope

Issues:
1. It's definitely not this cold in here
2. If I disconnect the I2C bus lines, it *still* gets data.  And that data changes (only slightly, but it's not the same)

adrianstevens, 2024-03-14:
@ctacke what's your hardware setup here? 

Its working normally for me with an isolated sensor

adrianstevens, 2024-03-14:
Can't repo - closing issue, feel free to reopen if needed

closed 2024-03-14 by adrianstevens