WildernessLabs/Meadow_Issues #484: Add docs for Meadow.Desktop

ctacke, 2024-02-21:
The new Meadow.Desktop needs documentation on the web site.  Developer docs are available here:
https://github.com/WildernessLabs/Meadow.Core/tree/develop/source/implementations/desktop/Meadow.Desktop

johnkattenhorn, 2024-04-19:
Even with these examples, I get a new Project to work for WinForms on Windows: 

Steps to Reproduce:

Using
Windows 11 Business
VS 2022 Professional

Steps to Reproduce:

1) New Project ---> Meadow Desktop
2) Fill In Project Details
3) Keep Target Framework at .net8.0 Default
4) Press F5
Get: System.Exception: You must run with a net8.0-windows target framework
5) Follow Instructions here (https://github.com/WildernessLabs/Meadow.Core/tree/develop/source/implementations/desktop/Meadow.Desktop
6) Rebuild 
7) Press F5 - Get: System.Exception: You must run with a net8.0-windows target framework

Also there are some additional items in the project file, I've tried removing them and keeping them but both result in the same error as mentioned in (7)

johnkattenhorn, 2024-04-19:
I only made the change from TargetFramework to TargetFrameworks to include both of the runtimes as per the documentation

johnkattenhorn, 2024-04-19:
This is caused by selecting `.net8.0` as Framework in the New Project Wizard - be sure to select `.net8.0-windows(preview) and it works as designed.