WildernessLabs/Meadow_Issues #491: Meadow.Cloud - Tokens expiring after an hour for OTA Updates

jorgedevs, 2024-02-27:
Working on OTA, leaving my app running for several hours, when I pushed an update from Meadow.Cloud, my Project Lab received the notification, but could not start to download the update:

```console
stdout> Wifi InvokeEvent NtpUpdateEvent returned CompletedOk
stdout> MQTT message received
stdout> Updater Message Received: 78de4dd00c1744a4aabba6b9fc6708e2
stdout> Update 78de4dd00c1744a4aabba6b9fc6708e2 is known but not retrieved
stdout> Updater State -> DownloadingFile
stdout> Device OS Version: 1.9.0.0, Update OS Version: 1.9.0.0
stdout> Attempting to retrieve https://www.meadowcloud.co/api/packages/78de4dd00c1744a4aabba6b9fc6708e2/download
stdout> Failed to download Update after 2 seconds: Response status code does not indicate success: 401 (Unauthorized).
stdout> Retrying attempt 1 of 10 in 1000 milliseconds...
stdout> Failed to download Update after 2 seconds: Response status code does not indicate success: 401 (Unauthorized).
stdout> Retrying attempt 2 of 10 in 1000 milliseconds...
stdout> Failed to download Update after 2 seconds: Response status code does not indicate success: 401 (Unauthorized).
stdout> Retrying attempt 3 of 10 in 1000 milliseconds...
stdout> Failed to download Update after 2 seconds: Response status code does not indicate success: 401 (Unauthorized).
stdout> Retrying attempt 4 of 10 in 1000 milliseconds...
```

It would still fail after multiple tries. The token to receive and downloads token should be refreshed or remain active for the duration of the app running.

doingnz, 2024-03-06:
@jorgedevs please see the Pull Request that I think resolves this issue here: https://github.com/WildernessLabs/Meadow.Core/pull/419

jorgedevs, 2024-03-06:
Thanks @doingnz, I mentioned your PR to the team, they're saying that that entire code region is being refactored and fixed.

doingnz, 2024-03-10:
@jorgedevs checking if the refactored and code has been committed or if that is still to be done? 

I ask as I noticed @adrianstevens made a commit on 9 March to add an exception handler `catch (MqttConnectingFailedException cfe)` and extra test `ClientOptions == null` in `ShouldAuthenticate()`

closed 2024-04-30 by lamebrain

doingnz, 2024-04-30:
@lamebrain what code change fixed the issue?

lamebrain, 2024-05-02:
@doingnz it was part of a big refactoring of the cloud service https://github.com/WildernessLabs/Meadow.Core/blob/develop/source/Meadow.Core/Cloud/MeadowCloudConnectionService.cs#L265