WildernessLabs/Meadow_Issues #494: Running `Cultivar` results in an OOM error about 30% of the time

bryancostanich, 2024-03-01:
**Describe the bug**
When I run the `Cultivar` app (or other apps that connect to the cloud), I get an out-of-memory err about 1/3rd of the time with output like:

```bash
stdout> making cloud log httprequest with json: {"eventId":10,"description":"device.health","measurements":{"health.cpu_temp_celsius":47.5347985347985,"health.memory_used":1518960,"health.disk_space_used":9052695,"info.os_version":"1.9.0.0","info.coprocessor_os_version":"1.9.0.0"},"timestamp":"2024-03-01T22:49:11.846142+00:00"}
stdout> memalign returned null when requesting 1048576 bytes (1048576 al
stdout> ignment) -- out of memory?
stdout> mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error 
stdout> code '142'" (142)
stdout> mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error 
stdout> code '142'" (142)
stdout> mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error 
stdout> code '142'" (142)
stdout> mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error 
stdout> code '142'" (142)
stdout> mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error 
stdout> code '142'" (142)
stdout> mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error 
stdout> code '142'" (142)
stdout> mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error 
stdout> code '142'" (142)
stdout> mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error 
stdout> code '142'" (142)
stdout> mono_os_mutex_trylock: pthread_mutex_trylock failed with "Error 
stdout> code '142'" (142)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Pull down the [cultivar](https://github.com/WildernessLabs/Cultivar/tree/develop), `develop` branch
2. edit your `wifi.config.yaml` file with your correct wifi credentials.
3. run the app in release mode (`meadow app run`).

**Expected behavior**
Should run without error.

**Meadow (please complete the following information as best as you can):**
Most of these vaues can be found by running `meadow device info` using the [Meadow CLI](http://developer.wildernesslabs.co/Meadow/Meadow_Basics/Meadow_CLI/).
 - Meadow hardware version: Project Lab v3.f
 - Meadow OS version: 1.9.0.0
 - If they are different, please provide these versions as well.
   - Meadow Mono version: v1.9.0.9
   - Meadow coprocessor version: v1.9.0.0

**Additional context**
This happens with other apps that connect to the cloud as well.

bryancostanich, 2024-03-14:
Turns out this is almost certainly caused by `System.Text.Json` and its massive memory overhead. We have removed it and replaced it with microJson, so this need to be retested with v1.next

bryancostanich, 2024-03-26:
This seems to be fixed in v1.10. I think it was the microJson switchover. Round Robin did not fix.

closed 2024-03-26 by ctacke

alexischr, 2024-04-01:
My initial profiling also seems to show that any crashes were not from a leak, but from overall high memory use of the app. I also seem to find that it doesn't crash now, but the memory use remains very high, at 26-27MB user heap usage (out of 29MB). At that point, a 1MB allocation (to extend the GC) with normal memory fragmentation will very easily be the first allocation failure.

I still suspect we have a leak in some apps, but I have switched to trying to just reduce memory and not looking for a bug, just like the microJson effort.